ixjf/MSIRGB #1: Doesn't work on MSI B350 Tomahawk (Nuvoton nct6795)

ShadowofCubes, 2018-11-23:
I can't use any commands. Even MSIRGB.exe -h doesnt work. But on https://github.com/nagisa/msi-rgb it is listed as working?

closed 2018-11-23 by ShadowofCubes

ixjf, 2018-11-24:
Does that mean it's working?

ShadowofCubes, 2018-11-24:
I just forgot to grant the programm admin rights. It's working on my Tomahawk B350. Thank you. Stupid msi shit.

ixjf, 2018-11-24:
Great, thanks!

I'm considering writing a GUI wrapper for this. Should be more intuitive to Windows users.

ixjf, 2018-11-30:
Hi. Please see the https://github.com/ixjf/MSIRGB again. I just commited a major change which includes a GUI. Shouldn't have a problem anymore with not having administrator rights as well.

ShadowofCubes, 2018-11-30:
That new GUI looks really nice. Could you please upload a compiled version of this? I don't have Visual Studio installed right now...

ixjf, 2018-11-30:
My bad. I forgot to upload the file. Duh! Here: https://github.com/ixjf/MSIRGB/releases/download/v1.0.0/MSIRGB-v1.0.0.zip

ShadowofCubes, 2018-11-30:
Thanks, it's really late here in Germany, I will look at it tomorrow.

ShadowofCubes, 2018-12-06:
Okay. I've found no issues so far, exept the high cpu utilization while using the "rainbow" script (I have 12 logical cores, so 10% is a lot): https://prnt.sc/lrnv99

ShadowofCubes, 2018-12-07:
Even if that script is empty or os.sleep() is set to 10000 the cpu utilization stays the same.
(When the script is empty it's even higher, about 25%)
The old Python script worked great and only used around 1%.
Maybe your programm is "starting" the script to often?[Sorry if I write absolute bullshit] Even if the script doesn't include a loop, it still runs forever like it's looped.

ixjf, 2018-12-07:
Right, I think I know what the problem is. The script service runs an infinite loop waiting for a shutdown signal for the script thread. There's no wait in between each execution, which is consuming too much time. Should be a 10 second fix, but I'll have to do that tomorrow morning.

ixjf, 2018-12-08:
Hi. Problem is fixed. See the latest release. Thanks for reporting!