ixjf/MSIRGB #6: Some red LEDs (EZ DEBUG LED?) are controlled, too

Nama, 2018-12-28:
I noticed, that some red LEDs are on on the MSI X470 Gaming Plus. One is directly on the one RAM I have installed and one is right at the PCI-E x16 port.

First I was confused, since everything works. EZ DEBUG LED from MSI isn't documented a single bit, so I wrote them a e-mail. The answer was, the EZ DEBUG LEDs are only turned on if the hardware is missing or a failure.

I started again your tool, turned off and on the RGB-LEDs... The red LEDs are controlled by your tool, too! They even flash if I set it in the tool.
So, maybe your tool changes too much on the MB?

Thats not really an issue, I just wanted you to know this behavior.

ixjf, 2018-12-28:
Right, it seems that the tool controls all LEDs on the motherboard. I wasn't sure about that. 
It's not that the tool changes each specific LED, but rather that the chip controls all the LEDs at once. Likely not something I could change if I wanted to.

Other than that, does the tool work properly?

ixjf, 2018-12-28:
Thanks for the info, though.

Nama, 2018-12-28:
It's also strange, that only the LEDs are turned on, where I installed something (GPU and **one** RAM).

> Other than that, does the tool work properly?
From what I tested, yes. I didn't test LUA scripts. I'm still on the initial release, since I can make it work with Python instead of LUA. Would have been nice, if you kept the command line still in the new version. A friend of mine uses the newest version and seems happy with it.

ixjf, 2018-12-28:
> It's also strange, that only the LEDs are turned on, where I installed something (GPU and **one** RAM).
Do you have any clearer picture of it? Your original one doesn't let me see the RAM LED you speak of. When it comes to the LEDs to the right, if they are at the back, then it's the backlight and is supposed to be on if MB lighting is on. A quick glance to photos of the MB led me to believe that's the only MB LED you have.

> From what I tested, yes. I didn't test LUA scripts. I'm still on the initial release, since I can make it work with Python instead of LUA. Would have been nice, if you kept the command line still in the new version. A friend of mine uses the newest version and seems happy with it.
The point of the new release was to make it simpler for the average user and also quicker to use, even for me. The only advantage you have with command line + Python option is in importing any module you want, or integrating my tool with other tools. Lua is pretty easy, similar in some respects to Python, and also simpler feature-wise, plus scripts will be loaded automatically on Windows start-up. You should give it a try.

Also, what MB is your friend using?

Nama, 2018-12-29:
> Do you have any clearer picture of it? Your original one doesn't let me see the RAM LED you speak of.

https://i.imgur.com/VlXosr8.jpg

> When it comes to the LEDs to the right, if they are at the back, then it's the backlight and is supposed to be on if MB lighting is on. A quick glance at photos of the MB led me to believe that's the only MB LED you have.

I mean these 4. They are the EZ DEBUG LEDs.
[image]

> The point of the new release was to make it simpler for the average user and also quicker to use, even for me. The only advantage you have with command line + Python option is in importing any module you want (debatable whether that actually gives much of an advantage), or integrating my tool with other tools. Lua is pretty easy, similar in some respects to Python, and also simpler feature-wise, plus scripts will be loaded automatically on Windows start-up. You should give it a try.

Ya, currently I am using just the example script, so I could use the example LUA script aswell. I just put the script in the Taskplaner of Windows.

> Also, what MB is your friend using?

Same as me, the X470 Gaming Plus.

ixjf, 2018-12-29:
> I opened the case just for you 
> They seem to be just for design and maybe they light up even if the official Mystic Light is used.
> https://i.imgur.com/VlXosr8.jpg
> 
They are simply for decoration according to this:
https://forum-en.msi.com/index.php?topic=305853.0
Could be that they give slot-specific failure information, but come to think of it, I had seen something about RAM LEDs while reverse engineering Mystic Light. To clarify, the RAM LEDs are controlled by the tool as well? They don't seem to have the same colour as the backlight, so is it that the tool turns them on/off but can't control their colour? But you say they flash too. What exactly can you change about them?

> I mean these 4. They are the EZ DEBUG LEDs.
> [image]
> 
Ah, okay then. These are debug LEDs only, indeed. The tool shouldn't affect them whatsoever.

Nama, 2018-12-29:
> They are simply for decoration according to this:
> https://forum-en.msi.com/index.php?topic=305853.0

Ya, seems like that.

> Could be that they give slot-specific failure information, but come to think of it, I had seen something about RAM LEDs while reverse engineering Mystic Light. To clarify, the RAM LEDs are [...]

The tool only decides if they are turned on or off and if they flash or not. Always red.

> Ah, okay then. These are debug LEDs only, indeed. The tool shouldn't affect them whatsoever.

Yes, I first assumed the deco-LEDs would be associated with them. But no, just blinky blinky.
If they also would be in RGB, I wouldn't have assumed that. I'd say, that's bad design by MSI.

That clarified a lot, thanks. You can close this, if you don't have any other questions.

Nama, 2018-12-29:
> Other than that, does the tool work properly?

I just noticed, that the newest version does like 7-8% CPU load. That's a lot! Even the Task-Manager says, that the power consumption is very high.
The Python script stays on 0-0,1%.

The first thing I wanted to do is, reloading the script. That's not possible, I need to kill the process in the Task-Manager.

ixjf, 2018-12-30:
Clicking on "Apply" or "Disable all lighting" stops the script. I suppose that's not the most intuitive thing. 
But the CPU usage problem had already been fixed. Are you using 1.0.2?

ixjf, 2018-12-30:
Nevermind, you're absolutely right. I fixed it in 1.0.1 and broke it again in 1.0.2. os.sleep is consuming all the CPU time.

ixjf, 2018-12-30:
Fixed in v1.0.3.

Nama, 2018-12-30:
Works perfect!

closed 2018-12-30 by Nama