ixjf/MSIRGB #7: Making AIDA64 control the LEDs?

Nama, 2019-01-12:
I am using AIDA64 to display hardware info's on the display of my Logitech keyboard. AIDA64 also supports changing the backlight LEDs of the same keyboard, i.e. depending on CPU temperature.

AIDA64 also supports graphic cards and motherboards, but with Mystic Light AIDA64 still wasn't able to control the LEDs. Is it possible for you, or would like to look into this, so AIDA64 can control also the MSI motherboard LEDs?

ixjf, 2019-01-12:
Is AIDA64 open source? If it isn't, I could at most do an hack. But even if I were to do that, it would still be missing support for a shit load of MBs. My tool only supports a select few. I _could_ make it work with others, but they work quite differently and would require some work. Plus, I don't have any of these to test the tool on.

Nama, 2019-01-12:
It isn't open source, sadly. The support of RGB-Hardware isn't documented either -.-
I don't want you to make support for all motherboards. Just make AIDA64 make use of your tool for the boards your tool supports.

An idea would be, that **your tool makes use of** AIDA64.
http://aida64.helpmax.net/en/external-applications/shared-memory/

Maybe make a list in your tool, live from the shared memory and make the user choose one entry. Maybe then the user can add steps on which value to which color to change.

Another idea would be... Maybe the best one. Just using LUA to access the shared memory. Would that be possible?

ixjf, 2019-01-12:
That page makes me think that the shared memory is purely for reading by external applications. So, you can't really change anything through it.
When I said I could at most do an hack on AIDA64, I was saying that you could inject some code on it. But that would require reverse engineering it, figuring out how to change the GUI, how to add the logic for the GUI, hoping they don't sue me, and hoping they don't release any new version which would essentially break my hack. So, not really doable :> 
The only reasonable way to do this is to create an entire new app that supports all that hardware. But that would require me to have the hardware to test it on. So, again, not really doable.

Nama, 2019-01-12:
I don't want you to extend AIDA64. I want your tool to use the read-only data in the shared memory to change LED colors.

Nama, 2019-01-12:
You would be able to change the LED color, autmatically, based on the values of your hardware.

- Fan speeds
- CPU temperature
- GPU temperature
- Board temperature
- Free space
- Memory usage
- and a lot more I assume

ixjf, 2019-01-12:
Oh, I get it. I thought you wanted to control the LEDs _via_ AIDA64.
I _might_ just do that. I'm currently working on another project and would rather not multitask, but give me a week or two and I'll look into this.

Nama, 2019-01-12:
Not gonna force or rush you.
Thank you for considering it :)

Nama, 2019-01-19:
I noticed that behavior only if I moved your tool to another location or updated it.

I tried just now what you told, but it opened again.

ixjf, 2019-01-19:
Thanks. Will have to look into this then.

ixjf, 2019-01-28:
Hi. I implemented support for querying AIDA64 info in the latest release, 1.1.0 beta. Would you like to give it a spin? I'd appreciate any suggestions of improvement - performance, API, you name it. You can find documentation for the new Aida64 module here: https://github.com/ixjf/MSIRGB/wiki/Scripts#aida64-module

ixjf, 2019-01-29:
I'm especially dubious of the performance of this. It's a terrible combination: WMI + exceptions + regex. However, when it comes to communication between the script service and AIDA64, WMI is the only way. Getting values from the registry won't work because AIDA64 stores data in a user account-specific key, while the service runs on a local administrator account (no access to HKEY_CURRENT_USER). Getting values from shared memory requires parsing XML, not much better. Exceptions - well, it's C#. Regex _might_ be replaceable but depends on how slow it is.

ixjf, 2019-01-29:
Great :) A call to IsInstalledAndRunning in the while loop would essentially be an extra call to GetSensorValue. I could probably make that slightly faster - I need to do some digging into WMI query syntax (it's SQL-like, so what I'm doing right now is SELECT * from the table, which is bound to be slower).

Nama, 2019-01-29:
Things like changing smoothly the color to more deep color while the value rises up would be nice, too. But that always limits to one or 2-3 static colors, which is lame.
If I come up with anything else, I'll make a PR with new scripts :)

Thanks

closed 2019-02-04 by ixjf

bpiero, 2020-03-06:
[nagisa/linux#102](https://github.com/nagisa/msi-rgb/issues/102)