ixjf/MSIRGB #8: How about adding support for MSI graphic card?

Nama, 2019-01-12:
Coincidentally, I have also a MSI graphic cards which supports Mystic Light (only the small Logo, but anyway).
I would appreciate it, if your tool can also control the LEDs of the graphic card and make them light the same way as the board.
I assume you don't own one, else you would have made that already...

If I'm correctly, the other "only graphics card"-tools use the original Mystic Light libraries. These wouldn't help you much. But maybe you get the idea...
https://github.com/Vipeax/MSI-LED-Tool
https://github.com/MinDBreaK/MSI-LED-Control

ixjf, 2019-01-12:
Coincidentally, I also have an MSI graphics card, but mine has no LEDs :)
The thing is graphics cards are not controlled via the MB chip. So it'd require adding basically an entirely new backend for the tool. And again, I have no GPU with LEDs to test the tool on, which makes it a bit difficult.

Nama, 2019-01-12:
> The thing is graphics cards are not controlled via the MB chip. So it'd require adding basically an entirely new backend for the tool.

Yes, I knew/assumed that.

closed 2019-01-12 by Nama