ixjf/MSIRGB #9: "Failed to load driver"

elelunicy, 2019-01-18:
So the last time I used MSIRGB a couple weeks ago it worked perfectly without a problem. And now it no longer launches it always says "Failed to load driver." What driver is it talking about? I have no clue what could have gone wrong.

ixjf, 2019-01-18:
It's talking about the driver MSIRGB uses to control the hardware. Are you using the same version as you had been before or did you upgrade since the last time? And which version is it?

elelunicy, 2019-01-18:
I was using v1.0.0 and it didn't work today. I then downloaded the latest v1.0.5 but it also has the same problem persist.

ixjf, 2019-01-18:
Could you open a command prompt as administrator and run the following command `sc query MSIRGB.Driver`? It will query the Windows' service manager and tell you whether the driver service is installed or not. That might give me a clue.

elelunicy, 2019-01-18:
```
[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.
```

Apparently it's not.

ixjf, 2019-01-18:
Could you replace your MSIRGB.DLL.dll with this one? Once you've done that, start MSIRGB again. It'll likely crash again. Once the tool closes, look for a 'debug_log.txt' in the folder where MSIRGB is and paste it here, please.

[MSIRGB.DLL.dll.zip](https://github.com/ixjf/MSIRGB/files/2774853/MSIRGB.DLL.dll.zip)

elelunicy, 2019-01-18:
```
driver .sys saved to system32 dir
driver service installed
failed to start driver service
driver service was not already running
start_drv failed
driver failed to load
```

ixjf, 2019-01-18:
Crap, forgot to log the important bit. Please try again.
[MSIRGB.DLL.dll.zip](https://github.com/ixjf/MSIRGB/files/2774874/MSIRGB.DLL.dll.zip)

elelunicy, 2019-01-18:
Now there is a new line "error: 183"

ixjf, 2019-01-18:
And everything else is the same?

elelunicy, 2019-01-18:
Yup

ixjf, 2019-01-19:
Just to make sure it's not some random problem - have you tried restarting the computer? If the error line appears along with the exact same output on the log file as before, the error code makes no sense at all.

elelunicy, 2019-01-19:
Yes I've restarted it multiple times and it's always the same. Here's the full log:

```
driver .sys saved to system32 dir
driver service installed
failed to start driver service
driver service was not already running
error: 183
start_drv failed
driver failed to load
```

ixjf, 2019-01-19:
Okay, try this DLL, then start MSIRGB again. It'll crash as usual. After that, open a command prompt as administrator and run `sc query MSIRGB.Driver`. If it doesn't say it's not installed (as it shouldn't), run `sc start MSIRGB.Driver` and send me the output.
[MSIRGB.DLL.dll.zip](https://github.com/ixjf/MSIRGB/files/2774916/MSIRGB.DLL.dll.zip)

elelunicy, 2019-01-19:
```
[SC] StartService FAILED 183:
Cannot create a file when that file already exists.
```

ixjf, 2019-01-19:
Does everything else on your computer work normally? I can't see this being an error from the driver itself (which I don't personally maintain), and it's clearly not a fault of MSIRGB since the driver service doesn't start from the command line. Did you do any Windows updates since the last time MSIRGB worked?

ixjf, 2019-01-19:
Just to make sure it isn't some weird ass bug related to file paths: search for 'registry editor' on start menu (just type it after opening the start menu), and replace the contents of the address bar at the top of the window that opens with 'Computer\HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Services\MSIRGB.Driver'. What does 'ImagePath' say?

elelunicy, 2019-01-19:
Yes my computer works as normal. Windows auto installed the cumulative 2019-01 update recently and that's about it. I also swapped a new CPU but that shouldn't affect drivers.

'ImagePath' says `\??\C:\WINDOWS\system32\drivers\inpoutx64.sys`

ixjf, 2019-01-19:
Looks about right. I have no clue why it's failing.

ixjf, 2019-01-19:
Any way you could run this tool? https://docs.microsoft.com/en-us/sysinternals/downloads/winobj
Start it, and choose "GLOBAL??" on the left. Look for a name "inpoutx64" or symlink "\Device\inpoutx64". Tell me if there is one.

elelunicy, 2019-01-19:
Yes it's there.

ixjf, 2019-01-19:
Does it still appear after you run `sc delete MSIRGB.Driver` on a command prompt with admin privileges?

elelunicy, 2019-01-19:
It's still there.

ixjf, 2019-01-19:
I might be on to something then. That shouldn't be there. No idea whether it was a bug in Windows or the driver, but that should disappear the moment the driver stops. I'll have to do some digging, but it's getting late, so I'll look into that tomorrow.

ixjf, 2019-01-19:
Driver objects should be deleted on restart, something I didn't consider yesterday. Meaning the driver initialization code must be executing. Weird.

elelunicy, 2019-01-19:
Could it be because something else I have is also using "inpoutx64"? It’s always there the moment after I log in Windows.

ixjf, 2019-01-19:
To answer your question, yes. Things would go wrong if the driver was being used elsewhere, since the driver tries to register the global object "inpoutx64" in the two instances. Let me write something quick to test that hypothesis.

elelunicy, 2019-01-19:
Yes it’s there before I run MSIRGB.

ixjf, 2019-01-19:
Try this. Tell me if it works. Debug log is in 'debug_log.txt', though if it works I won't need it.
[MSIRGB.DLL.dll.zip](https://github.com/ixjf/MSIRGB/files/2775519/MSIRGB.DLL.dll.zip)

elelunicy, 2019-01-19:
It worked. Thanks!

ixjf, 2019-01-19:
Great! I'll push the fix officially asap then.

closed 2019-01-19 by ixjf

ixjf, 2019-01-19:
Could you test this and tell me if it's working fine for you before I publish it?
[Release.zip](https://github.com/ixjf/MSIRGB/files/2775827/Release.zip)

elelunicy, 2019-01-19:
Yes it works.

commit df696cbb (2019-01-19): Fix #9 (if some other program were using inpoutx64, MSIRGB would fail to load the driver)
Fix MSIRGB not keeping track of open handles to the driver service, leading to incorrect stopping/deletion of the driver service and not allowing MSIRGB to start afterwards
Add code for safety regarding starting/stopping the driver service
commit d7838409 (2019-01-19): Fix #9 (if some other program were using inpoutx64, MSIRGB would fail to load the driver)
Fix MSIRGB not properly keeping track of open handles to the driver service, leading to incorrect stopping/deletion of the driver service and not allowing MSIRGB to start afterwards
Add code for safety regarding starting/stopping the driver service