ixjf/MSIRGB #10: MB testing.

Zifoduk, 2019-01-27:
Worked fine with the MSI A320M bazooka. However with MSI gaming center installed, causes conflict. Otherwise, without it installed works fine

ixjf, 2019-01-28:
What do you mean by "causes conflict"? What happens?

Zifoduk, 2019-01-28:
It will flicker between the Msi gaming centers led settings and this
applications led settings rapidly. It also will refuse to turn off the
Gaming centers LED Settings until this application is turned off first.

ixjf, 2019-01-28:
Right. You have to choose one tool - can't have both running at the same time. 

You might be able to keep MSI's tool installed if you set LEDs disabled on it, though that's only a guess, and it may very well not be true. Another possibility, if you don't mind messing around, is setting MSI's background service as disabled. You won't have to uninstall the tool, and it won't mess up anything. For that, go to the start menu, type "services.msc", find "LEDKeeper" (again - I'm guessing here, I'm not sure MSI Gaming Center works in the same way as MSI Mystic Light, but it probably does - it'll do you no harm to look for the service anyway), right-click on it, click "Properties", set "Startup type" to "Disabled", click "Apply", click "Stop" if the button is enabled, "Ok", done. Repeat and set as "Automatic" if you want to reenable it.

Regarding it refusing to turn off MSI's settings - there's no connection between my tool and MSI's, so I find that very weird. Maybe MSI's tool turns on if the LEDs are enabled. Shouldn't have a problem with that if you follow the steps above.

zu6fx, 2019-01-31:
Just to add on, tested the application on MSI B250M Mortar and it also works fine.

ixjf, 2019-01-31:
I'm closing this issue. If you still encounter problems, please open a new issue.

closed 2019-01-31 by ixjf