ixjf/MSIRGB #14: Hue Wheel script using 0.5-1% CPU time

ixjf, 2019-02-23:
On a Ryzen 5 1600 @ 3.7GHz. Way too much time for what it's doing.

Additionally, check if AIDA64 module functions aren't taking too much time as well.

SimoDax, 2019-02-23:
Took a quick look, I may be wrong but seems like the driver's fault, lua side seems ok
[image]

SimoDax, 2019-02-23:
Dug a bit deeper, kernel32 is really taking it calm and slow..
[image]

ixjf, 2019-02-23:
The problem is really how much time the script thread is paused for between colour steps. I have set a delay of 20ms, which is clearly too small.

SimoDax, 2019-02-23:
Isn't it 80?

ixjf, 2019-02-23:
It is, indeed. I thought I had it at 20ms yet. I've got to dig into this some day, though increasing the delay will make the transition less smooth/slower.

SimoDax, 2019-02-23:
Since system calls are taking up so much, is set_led_enable(true) needed at every colour change? Half of CPU time goes into that alone, but maybe it needs to be "refreshed", I don't have fully understood how the SuperIO chip works yet
https://github.com/ixjf/MSIRGB/blob/b2dec3b4d6844865cc22b00c036bde5729d83ebf/MSIRGB.DLL/logic/Sio.cpp#L97-L103

ixjf, 2019-02-23:
Sort of. It is not necessary for set_colour to work, it is there (and in every other set call to the chip) so that as long as a script is running, some third-party won't turn off the LEDs.

ixjf, 2019-02-23:
How much faster is it if you comment out all calls to set_led_enabled?

SimoDax, 2019-02-23:
I haven't recompiled, I wrote 20% by looking at the profiler report (the 2nd I posted)

Additionally, 33% goes into read_uint32, which in the specific case of hue wheel is unnecessary since all indexes are overwritten - maybe set_colour could be overloaded with a version without index that loops by herself through 1 to 8 and hasn't calls to read registers?

ixjf, 2019-02-23:
You necessarily need to read cells from the chip to change a colour - you can't read any less than one byte. Also, it sounds like these calls simply take a lot of CPU time because, well, they are called a lot of times. They're the gist of the script.

Also, while set_colour _could_ be written in some other way such that it doesn't read the entire dword a time, it's not really the problem since the script was already taking its time before I changed the function to work that way (try previous release to confirm) - it previously only read one byte from each dword.

ixjf, 2019-02-23:
In fact, as you can see in the report, the largest amount of time is really in the Lua VM, which implies the delay is too small and the loop is running too often.

ixjf, 2019-02-23:
The report doesn't really give much useful information here. Some calls are obviously going to be taking the largest amount of time because they're called so often. A more reliable way to see which functions are taking too long would be to see how much time _one call_ was taking.

SimoDax, 2019-02-23:
> You necessarily need to read cells from the chip to change a colour

Even if you want to overwrite everything?

> A more reliable way to see which functions are taking too long would be to see how much time one call was taking (or an average).

The report is indeed averaged over 11 seconds. I agree with you, the increased load comes from calling set_color very often. but I'm instead focusing on decreasing the cpu time consumed on each call rather than lowering the frequency

ixjf, 2019-02-23:
The thread stays asleep for more or less the same time, it just does less work per iteration.

I just tested it and what's really taking too much time is the 8 calls to SetColour per iteration.

SimoDax, 2019-02-23:
> is the 8 calls to SetColour per iteration.

And we go back to the system call overhead
Tomorrow I'll try some of my ideas and report how they work out

ixjf, 2019-02-23:
The best I can do about the set_led_enabled calls is forcing them to be called by the script instead, but I don't think that makes much sense since the user shouldn't have to be bothered to take into account potential third-party programs changing the hardware at the same time. Either way, the time it saves is as much as I calculated, 0.1% or less.

ixjf, 2019-02-23:
https://imgur.com/a/emGYTuP

Still 25%, but 0% CPU usage.

ixjf, 2019-02-24:
See 690c2c3. I managed to reduce the CPU usage down to 0.1-0.2%.

SimoDax, 2019-02-24:
> I managed to reduce the CPU usage down to 0.1-0.2%
 
It wasn't the _only_ call taking up time, you reduced to 1/4 the bit reads and writes, of course it drops the usage. My suggestion was to skip the reads altogether when setting all nibbles

> Still 25%, but 0% CPU usage.

How did you reduce the number of calls to set_colour? 195*8 gives the old usage, but  the `for` loop is still present in the script

SimoDax, 2019-02-24:
In order to get accurate usage, I edited the script like this: 

    local x = os.clock()
    local s = 0

    for i = 1, 8 do
        Lighting.SetColour(i, r, g, b)
    end

    print(string.format("elapsed time: %.3f\n", os.clock() - x))

This is just by commenting set_led_enabled, roughly 20% less time needed. Task manager usage goes down accordingly as well. (Please note that I'm still using the old set_color version, uint32 calls)
[image]

ixjf, 2019-02-24:
Of course it isn't the only call taking time, but any other option is not viable. Adding a SetAllColours function might reduce the CPU usage in this script, but it won't in some other that might want to set different colours. It's still an option, but right now I'm not really willing to put time into this.

SimoDax, 2019-02-24:
> Adding a SetAllColours function might reduce the CPU usage in this script

Just tried it, the usage drops well below system resolution:  
[image]

I think an overload would be fine: If you want to change all colours just call it without index. I can make the modification myself and pull request if it's ok for you

ixjf, 2019-02-24:
What's the C++ code like? And which version of MSIRGB are you using?

ixjf, 2019-02-24:
If you're simply removing the loop, obviously you're going to get a lot less time. But a SetAllColours function is going to have 8 times the driver calls.

ixjf, 2019-02-24:
I mean, I could check it out, but I'm _pretty_ sure there's no way to access_update more than once cell at any one time.

ixjf, 2019-02-24:
Unless you're still using the old version with the uint32 calls, in which case that's a more reliable comparison?

SimoDax, 2019-02-24:
> Unless you're still using the old version with the uint32 calls

Yep, Version b2dec3b4d6844865cc22b00c036bde5729d83ebf

The code for now is a stupid if added on top beacuse i just wanted to check the performance

[image]

The `for` in lua script is removed, instead I just call `Lighting.SetColour(0, r, g, b)`

ixjf, 2019-02-24:
I've thought about batching calls - it could fix this issue without having to actually modify the script. Essentially, batched calls are stored and then executed all at once, so you could optimize, skipping all the unnecessary calls. Maybe more trouble than it's worth, though.

ixjf, 2019-02-24:
Yeah, that should work. However I added a fix for #13 along the way in yesterday's commit, so you'll necessarily have to use the latest code :>

ixjf, 2019-02-24:
Also, preferably, SetColour would be passed nil rather than 0, and the Lua binding would check that and pass null to the DLL, which would then be converted to an std::optional<std::uint8_t>.

SimoDax, 2019-02-24:
Oh, thank you, unluckily I don't know typecasting between Lua and C++. I just edited the binding to include 0 :)

ixjf, 2019-02-24:
I'm thinking batching calls might be a better idea than changing only SetColour. It'd be more general purpose and doesn't require changing the Lua function at all. Opening a new issue for that.

ixjf, 2019-02-24:
Thanks for the work on this, by the way!

commit 9f30e84e (2019-02-24): Implement #17
Fix #14
closed 2019-02-24 by ixjf