ixjf/MSIRGB #17: Add call batching?

ixjf, 2019-02-24:
Would fix #14. Something along the lines of Lighting.BatchBegin() and Lighting.BatchEnd(). Calls would be stored and applied only when BatchEnd is called. Would allow for removing unnecessary overhead of Lua and unnecessary set_led_enabled calls.

ixjf, 2019-02-24:
Batching would be optional. Essentially, there is a batcher class such that if BatchBegin wasn't called, commits requested change immediately, else stores it. For set_colour, the code in #14 would be a good start.

SimoDax, 2019-02-24:
Can do it
what about the batcher class, should it be member of Sio?

ixjf, 2019-02-24:
I just implemented this in commit 9f30e84. CPU usage is now 0% even with the previous delay of 80ms.

closed 2019-02-24 by ixjf