ixjf/MSIRGB #18: Lighting.SetColour should take in a nibble-size value for colour

ixjf, 2019-02-24:
Colours have 12-bit depth. Lighting.SetColour should take in a nibble-sized colour value for each channel, rather than making approximations.

SimoDax, 2019-02-24:
By approximation you mean bitmasking to 0 the 4 least significant bits of the byte values?

ixjf, 2019-02-24:
It divides 16-bit colour value by 16, so you get a single-byte value. It's basically what Mystic Light does.

ixjf, 2019-02-24:
It's the same as what you're saying, except the value is then shifted right.

SimoDax, 2019-02-24:
Hm, this and #19 seem hard. Some function futher up will need to make that approximation if SetColour isn't doing it. Unless you present the user only with the actual colors he can set. But then again what about the scripts?

ixjf, 2019-02-24:
This one should be pretty simple. It's a matter of removing the approximation and making the SetColour Lua binding and Lighting::set_colour fail if an invalid colour is used. Of course, this is a breaking change, so this would be 2.x.

#19 is more complicated. It'll require rewriting the colour picker XAML component.

SimoDax, 2019-02-24:
> making the SetColour Lua binding and Lighting::set_colour fail if an invalid colour is used

If color space conversion from hsv to rgb is use, how can one know which HSV tuples values will yeld valid RGB nibbles?

ixjf, 2019-02-24:
You can do the approximation in the script, or simply range over only RGB values where R = 0x00, 0x11, 0x22, 0x33, etc, and same for G and B.

commit 3b750c98 (2019-02-27): Fix #18
closed 2019-02-25 by ixjf