ixjf/MSIRGB #19: Colour picker should be changed to display only the 3375 available colours

ixjf, 2019-02-24:


SimoDax, 2019-02-24:
In case it helps, I generated a palette with only the 3375 pixels that are settable without approximations
Empty rows and columns could be deleted, the remaining spaces filled with the closest neighbour colour, and used with the current picker code
[image]

Another option would be to set the color inserting 0-15 values in text boxes and having the picker rectangle display a preview

ixjf, 2019-02-25:
I would prefer if there was a colour picker. Still not sure how to do this. Probably will leave it like this for now.

SimoDax, 2019-02-25:
This makes me wonder: what about the motherboards MSI sells stating "16 million colours"? Smells like burnt..

ixjf, 2019-02-25:
As far as I know, all the motherboards that MSIRGB supports are advertised as having only 7 choices of colour. Those that do say 16 million colours all likely use a separate Renesas chip for LED control.

ixjf, 2019-02-26:
Fixed.

commit 1d54329a (2019-02-26): Fix #19
commit 410b6a24 (2019-02-26): Fix #19
commit c013f981 (2019-02-26): Fix #19
closed 2019-02-26 by ixjf