ixjf/MSIRGB #25: Color goes to white between each change

thcedi, 2019-03-28:
I've got a B450 Tomahawk, the tool works great except that between each color change the color is changing to white. So it's like color1 - white - color2 - white - color3 - white and so on. Is it supposed to be like that?

ixjf, 2019-03-28:
It isn't supposed to do that _on supported motherboards_. I'm afraid the B450 TOMAHAWK mobo is not supported.

closed 2019-03-28 by ixjf

LunaMoo, 2019-04-07:
Hi, I have the same mobo and it is actually supported, however you first have to run MSI Mystic Light software and turn the RGB off in there, otherwise the effect selected there is causing issues to effects from this utility:3.
@ixjf you might want to add a note to the readme about it. Anyway thanks, your app allows for much smoother control than the Mystic Light app.

ixjf, 2019-04-08:
By _supported motherboards_ I was referring to the list of motherboards that are on the README in the 'Supported motherboards' section. As it says on the README, it's entirely possible the LED control works the same as in those but is not one of the motherboards I can (kind of) guarantee the tool will work properly on.

I'd expect that disabling MSI Mystic Light is an obvious thing to do before attempting to run MSIRGB. Perhaps alternatively I'll add a check to MSIRGB to warn the user if Mystic Light/Gaming App is on.

Anyway, so the tool works properly on the MSI B450 TOMAHAWK then?