ixjf/MSIRGB #35: MSI B450-A PRO

BIGV10, 2019-05-04:
MSI B450-A PRO is working. but manufacturer instead of "Micro-Star International Co., Ltd." is "Micro-Star International Co., Ltd" (without last dot)

[L"Manufacturer"] = L"Micro-Star International Co., Ltd"
[L"Product"] = L"B450-A PRO (MS-7B86)"
[L"Version"] = L"2.0"

but color palette is strange. If i need to set green color then i need to set R=f, G=0, B=f, pumpkin script is kinda blue (#00e0ff), strobe script is kinda yellow (#e09000)
Police script is nice red and blue,

BIGV10, 2019-05-05:
plus after application start, MSI Command Center stops to recognize temperatures and RPM of fans.

ixjf, 2019-05-05:
I can't confirm right now but that motherboard probably works differently. Adding support for it, like for many others, is technically possible and not even difficult, but since I won't be able to test, I think it's best to simply refrain from doing it. 

Did everything go back to normal?

tmfadhlul, 2019-05-05:
it not work on my msi b450 a pro i cant even install it

tmfadhlul, 2019-05-05:
@BIGV10 so, I need to change the Manufacturer and rebuild the MSIRGB.dll ?

BIGV10, 2019-05-05:
> Did everything go back to normal?
If i terminate `MSIRGB.exe` and `MSIRGB.ScriptService.exe` and relaunch MSI Command Center, it works properly. If i launch MSIRGB after MSI Command Center, Command Center will function properly.

BIGV10, 2019-05-05:
> Did everything go back to normal?

If i terminate MSIRGB.exe and MSIRGB.ScriptService.exe and relaunch MSI Command Center, it works properly. If i launch MSIRGB after MSI Command Center, Command Center will function properly.

BIGV10, 2019-05-05:
> @BIGV10 so, I need to change the Manufacturer and rebuild the MSIRGB.dll ?

You can change Manufacturer and add `L"7B86"` to `static const std::list<std::wstring> supported_mbs` and rebuild MSIRGB.dll
or
You can disable check for motherboard here `private void TryInitializeDll(bool ignoreMbCheck = false)` (change `false` to `true`) in MSIRGB.GUI -> MainWindowModel.cs and rebuild MSIRGB.GUI (this one is not recommended i think)

ixjf, 2019-05-05:
Neither options are recommended. There's a reason why the motherboard isn't listed as supported. Unfortunately, unless someone is willing to maintain support for other motherboards, I can't really do anything here.

closed 2019-05-05 by ixjf

diogotr7, 2019-11-22:
I have this motherboard and ran into the exact same issues. How would i go about making it work correctly? Would i need to analize packets (?) sent by MSI mystic light somehow? 

The end goal is to hopefully integrate MSIRGB.fll into [Aurora](https://github.com/antonpup/Aurora).

diogotr7, 2019-11-23:
I was able to make it work correctly and send the correct colors using a **very** ugly hack: 

Since before anyn changes the colors displayed were these:
```
black -> white
white -> black

red -> cyan
green -> magenta
blue -> yellow

yellow -> blue
magenta -> green
cyan -> red
```
, i figured if i just flipped all the bits it would correct them, so that's what i did here:
in: 

````
void Lighting::batch_commit()
{
std::uint8_t r = (r_cell & (0x0F << (nibble_pos * 4))) | ((0xff - colour.r) << (!nibble_pos * 4));
std::uint8_t g = (g_cell & (0x0F << (nibble_pos * 4))) | ((0xff - colour.g) << (!nibble_pos * 4));
std::uint8_t b = (b_cell & (0x0F << (nibble_pos * 4))) | ((0xff - colour.b) << (!nibble_pos * 4));
}
````
Notice the ````0xff - color``` . Is there a better way to integrate this board? 

Keep in mind that i didnt test very well all the other values, although interval, flashing, and breathing seemed to work just fine.

diogotr7, 2019-11-23:
I've found that adding ````L"7B86"```` to the reversed devices list from #40  seems to work fine.

ixjf, 2019-12-02:
If it works, it probably works. I reverse engineered Mystic Light to figure out how each particular motherboard worked, so that's what I'd recommend you to do if you'd like to confirm. The logic is in MBAPI_x86.dll. The DLL exports an initialization function and you can trace it back to the main LED logic.

ixjf, 2019-12-02:
If you find that you've got it correct, send in a pull request and I'll try to merge it when I can (can't promise that will happen any time soon, though, sorry!).