ixjf/MSIRGB #37: B450 Gaming Plus

Mouse518, 2019-05-07:
MSIRGB does work with the B450 Gaming Plus, but you have to take into account the codes will do the opposite of what you want them to. To make red for instance, the colors must be set as R 0x0, G 0xf, B 0xf.

ixjf, 2019-05-07:
What about everything else? Does everything work properly other than that?

ixjf, 2019-05-14:
I'm closing this due to inactivity.

Guru-Guru, 2019-05-18:
I have the microATX version (B450m), same thing happens. I'm guessing that there's not much difference between the two but for what is worth, everything else seems to be working properly.

Mouse518, 2019-05-18:
Everything else works as expected, including everything in the examples download.

ixjf, 2019-05-18:
I'll look into it then but can't promise when.

ixjf, 2019-08-27:
I'm closing this as there are already two pull requests that tackle this issue. Thanks for reporting.

closed 2019-08-27 by ixjf