ixjf/MSIRGB #38: MSI Z270 M7 not recognized

Martinocom, 2019-05-10:
Hello!
My MoBo is on the supported list, but the program won't launch. It says to report this, so here I'm. Computer is fresh, just formatted.
[image]

ixjf, 2019-05-10:
Could you run this command on the command line and tell me the output?

```
wmic path Win32_BaseBoard get Manufacturer
```

Martinocom, 2019-05-10:
Here it is! (no admin priviledges)
[image]

ixjf, 2019-05-10:
Does it work with this DLL?

[MSIRGB.DLL.dll.zip](https://github.com/ixjf/MSIRGB/files/3168161/MSIRGB.DLL.dll.zip)

Martinocom, 2019-05-10:
Worked, but shouldn't it have something more, like colours?
[image]

ixjf, 2019-05-10:
Alright, thanks, publishing the fix asap.

closed 2019-05-11 by ixjf

ixjf, 2019-05-11:
By the way, can I add your motherboard to the list of verified working motherboards?

Martinocom, 2019-05-11:
Sure! Happy to contribute :)

PS: Is there a way to set one colour without cycling? Like a "static blue"? I currently have all 8 colours set as same blue to achieve that.

ixjf, 2019-05-11:
That's the only way to do it, really.

Martinocom, 2019-05-11:
Np, thx for fixing it so quickly. To add some details, all RGBs around the mobo and also the led strip connected to the 4-pin RGB works perfectly.

Happy coding.

commit 49ab90ec (2019-05-11): Fix #38