ixjf/MSIRGB #39: Add B450M Bazooka Plus to supported devices list

FadedCoder, 2019-05-29:
This works perfectly with the motherboard (https://www.msi.com/Motherboard/B450M-BAZOOKA-PLUS.html).

Just the starting not supported message is annoying every time I open it. Also, is there any reference for the API? I'm planning on making a universal app for MysticLight, SteelSeries Engine and other lighting apps.

FadedCoder, 2019-05-30:
My bad, everything is not as intended. The colors are reversed. (0xF00 is 0x0FF). Will commit a push fixing it soon.
Just having trouble opening the project on VS 2019.

ixjf, 2019-05-30:
> Also, is there any reference for the API?
What API are you talking about?

ixjf, 2019-05-30:
Also, this is #37.

FadedCoder, 2019-05-30:
Gaming Plus is a different motherboard. I am submitting a PR for this motherboard as soon as I manage to compile it from VS2019. WHY CAN'T VISUAL STUDIO WORK FOR ONCE.

By API I meant like a reference document for the DLL. Nevermind that, the code is self explanatory.

FadedCoder, 2019-05-30:
@ixjf have a question. Does the 8 colors switch randomly? Or in some order?

ixjf, 2019-05-31:
It is a diferent motherboards but the behaviour is the same.

The colours change in the order from index 1 to 8.

FadedCoder, 2019-05-31:
I was thinking of that too, but in my motherboard it goes from 1 to 8, then 8 to 1.

FadedCoder, 2019-06-01:
@ixjf when I try to run motherboard support checking (the WMI connect part) on another app, I get an error access denied. But this does not happen on the original app for same code. How did you fix this?

ixjf, 2019-06-09:
> I was thinking of that too, but in my motherboard it goes from 1 to 8, then 8 to 1.

I think that's correct.

> @ixjf when I try to run motherboard support checking (the WMI connect part) on another app, I get an error access denied. But this does not happen on the original app for same code. How did you fix this?

MSIRGB runs as administrator, that's probably why.

ixjf, 2019-07-05:
Closing this as you already have a pull request open.

closed 2019-07-05 by ixjf

bobalob, 2019-08-01:
Hey,

Not an issue since this is not a supported board, but just wanted to let you know this is working on a B450M Moratar Titanium board, however the RGB values are reversed. 

To get full red you need to set R=0, G=F and B=F. 
To get Yellow, R=0, G=0, B=F.
etc...

I **think** the B450M Mortar Titanium is just a B450M Mortar with a different paint job.

Thanks for taking the time to work on a sweet utility, even with colours reversed it's still way better than Mysic Light! 👍

ixjf, 2019-08-27:
Thanks! It is most definitely an issue. I'll add it to the list of supported motherboards with a note.

2.2.1.3 should fix the reversed colours issue, though I can't tell you when that will be ready (can't work on MSIRGB any time soon due to university).