ixjf/MSIRGB #43: Apply effects on Windows boot

Martinocom, 2019-06-12:
Hello! My all harwdare is connected to one power strip, that I shut down every time I turn off all my things off (PC/printer etc.). As it, the MSI monkeys doesn't store RGB info in some kind of memory and when all is cut of from power supply, it simply forget the settings and restore default RPG settings (red light). _Yes, date is ok and everything in BIOS is saved even if I shut down all the things._

This happened also with the official app: when started, all lights went red, then after login on Windows it will be set on blue (as I wish).

Is there (or it will be) a possibility to start the MSIRGB app on startup and apply automatically effects I want?

ixjf, 2019-06-12:
Settings applied through the GUI are not applied on startup, but scripts are. You can write a simple script to apply the settings you want. You can look at the existing scripts and the wiki (https://github.com/ixjf/MSIRGB/wiki) for guidance.

Martinocom, 2019-07-02:
Hello! It's me again. Unfortunately, not always my script will launch at Windows start. I'm not sure WHY this happens, but right now I turned on my PC and script isn't running. Even opening the MSIRGB app will not work: still, the "strobe" script is there marked as "working".

Here is my pc (should be all blue, not default red):
https://i.ibb.co/Kr85W6k/dont-work.jpg

Here is the app, opened after boot:
https://i.ibb.co/v37HZ69/dont-work.png

The most interesting thing of this is that, if I put my pc on "sleep mode" (or hybernate), the script WILL actually work if I turn the PC on again. Every time.

ixjf, 2019-07-02:
I use it myself and it works fine. Maybe there is some incompatibility with your motherboard, as from the GUI it doesn't seem to be reading the info about the LEDs correctly. What motherboard do you own?

ixjf, 2019-07-04:
When this happens, do you see a service "MSIRGB.ScriptService" running?

Martinocom, 2019-07-05:
Just turned on and checked my PC.
Service is running and log tells that the script was initialized and finished. There is not timestamp tho.
https://i.ibb.co/DpxWCQL/doesntwork.png

ixjf, 2019-07-05:
Right, I forgot log doesn't have timestamps. Can I see your script? I think I have a general idea of the cause already.

Martinocom, 2019-07-05:
`
local colors = {["r"] = 0x0, ["g"] = 0x0, ["b"] = 0xf}

Lighting.SetFlashingSpeed(0)
Lighting.SetBreathingModeEnabled(false)

for i = 1, 8 do
    Lighting.SetColour(i, colors.r, colors.g, colors.b)
end
`

The code is based on Strobe.lua (that doesn't work either in "auto-start" mode).

I must also correct myself about script that work after hybernate: it stopped to work too.
Just for debug purposes, I deleted the log file and neither at start, not reboot it will not appear (so I think script parsing procedure doesn't even start). Disabling and re-enabling it in app worked perfectly and creates the file.

ixjf, 2019-07-05:
Just to confirm my suspicions, does it work with the Hue Wheel script?

Martinocom, 2019-07-05:
Yup, after applying it manually, it's doing exactly the same thing as in your preview (here: https://github.com/ixjf/MSIRGB/raw/master/media/hue_wheel.gif )

ixjf, 2019-07-05:
Upon Windows startup?

Martinocom, 2019-07-07:
To make a point of what is happening, I'll make a sort of list of behaviours.

# Lights are not ok when
I think a major problem of this all is that I fully unplug the power strip after shutingdown my PC. It seems to me that values of lights (eventually scripts) are stored in some "volatile" memory, that reset to default when MoBo isn't powered-up. Steps to reproduce:
1) Open the app and set up the script/lights,
2) shutdown the pc,
3) unplug power strip,
4) wait some time, just to be sure all leds inside PC and other hardware are gone
5) again, power up everything,
6) turn on pc (it will load default colors)

At this point it doesn't matter if I shut it down again, put it on hybernate/sleep or others. Lights will be just default.

And here should came your app: after a full boot it should change the color. This is not happening, for some reason. As you pointed out, maybe _chip is not ready_: I remember the MSI app applied color settings after 3-5 seconds passed from when the login screen appeared. (like that: _boot - login screen on W10 - 5 seconds - colors applied_)

**SOMETIMES** when I open the app only to check if script is _"checked"_ (but I don't touch anything, no settings are changed), and then, when PC is put into hybernate/sleep, the script will eventually launch and lights will load. This is not happening always, since I tested it more carefully, and I didn't find a univocal way to make it set colors correctly. Sometimes it will launch, sometimes not.

# Lights are ok when
The lights will be ok only if I set them (via script or just setting up colors) and NEVER AGAIN unplug the power strip. I can do anything: shut down, sleep/hybernate mode. Just literally all the things, but at every startup lights will be all fine, just like I want them to be. After unplugging the power strip, all the magic is gone.

# Conclusions
**No**, I can't just "don't unplug anymore". There are some leds on the MoBo that are making my room "not-so-dark" at the nights, and I'm very _susceptible_ to lights at night time. 

So mabye there is no solution for **all** this problems. I remember that with the official app was happening the same thing. I could set up all colors I want, but if unplugged, it will start red (default) and eventually, after windows boot, will change to desire colour. 

I remember that before right color was applied, it will flash at least 2 times: one turning off and on the default ones. Immediatly after that, turning off the default and turning on my colors. All this process lasts for less than 1 second. With your app, when colors are applied, this happens only ONE time, lights just turn on to desired ones.

ixjf, 2019-07-07:
Colours are not saved between shutdowns, whether using the MSI app or MSIRGB, so that's normal. The difference is that the MSI app resets the colours on a regular interval, like the Hue Wheel script, while the other effects are run once only. It seems that something is not allowing me to set the settings so soon.

ixjf, 2019-07-07:
Actually, if you say it works fine as long as you don't unplug the power strip, then I'm probably missing something in the initialization code for your motherboard. I turn off my power supply all the time and it still works, but I'll do some testing of my own now.

ixjf, 2019-07-07:
Can reproduce but not sure what the problem is. At first glance, doesn't seem to be a problem with actual hardware-related code, but will have to take some time some other day to look into it.

Martinocom, 2019-07-07:
Np, it's not a crutial thing. I'm just collaborating at this project. 
Let me know if you need to know something else.

Martinocom, 2019-07-12:
Just to add some details, today happened something REALLY strange.

My PC was on the right point to shutdown (the screen went already black), but just one second before all lights should turn off, ALL of them turn blue (from red), remained there for like 1-2 seconds and then all PC shut down like nothing.

After 1-2h, when I decided to run it again, all lights were still blue. I didn't uplug it from the wall, neither opened your app.

Ah, to add something, my MoBo is taken from Amazon Warehouse Deals store; maybe has the chip "messed up"?

ixjf, 2019-08-27:
Unlikely.

Lights turning blue-ish when turning the PC on/off is normal. It's the default colour when the chip is reset.

I can't work on this now (MSIRGB in general), or any time soon, but I'll leave this open for when I can.

ixjf, 2019-08-27:
To summarize: MSIRGB fails to change LED settings at Windows start-up when coming from cold boot. Works with Hue Wheel script and any other script that resets the settings at a regular interval, so it's clearly something about the timing of when MSIRGB sets the settings the first time on start.

Can reproduce this, just don't know why it happens.

Martinocom, 2019-08-28:
Yes, all confirmed, if _"cold boot"_ means that I must turn on the electric strip + PC.

On "cold boot", no matter what are the settings, the colour will be always default (red). There is like 0.01% chance that the colour will be right, but from then I didn't see this happen.

Otherwise, if the power strip **is** already on, _"things"_ could happen:
- it can just stay at it is after every boot (=default),
- it can just switch to the right color on the boot (=blue, like 20% of chance),
- it can be put on "hybernate/sleep" mode (default colour still on) and after waking up it could (not always, like 20% of chance) switch to the right colour (=blue).

If interested, I discovered that the same thing happens with my Corsair K70 keyboard. It not always get triggered by iCue: I must unplug/plug it again (or move left-right a little switch behind the keyboard) to make the trigger work. The funny thing is that the keyboard **IS** actually recognized by Windows, but it's not from the iCue perspective.

And just a reminder: my MSI Z270 is an Amazon Warehouse one, it could be somehow "broken" but still working. For example, my ice-blue colour on the MSI is getting more "green" than "blue", and some of the LEDs are completly green when set to that color. This happens with your and the official app, and since it's not crutial, I'm not worried about it.

ixjf, 2019-08-31:
Don't worry, if I can reproduce this, then it's definitely not your motherboard. Unfortunately, my PC is currently dead (RMA'd the SSD) and university is starting very soon, so this could go unsolved for a while.

As a temporary workaround, you could write a script/edit the existing ones to apply the changes 1-2 seconds later. That should make it work properly. Something like this: `os.sleep(1500)` at the very top of the script. It will pause the script for 1.5 seconds, should be enough.

Martinocom, 2019-09-02:
Unfortunately, pausing the script with `os.sleep(1500)` didn't work. Neither if changing this value in range from 1000 to 4000. As indicated, I put this line in the first line of the script. If lanuched directly from the app, it works perfectly, with specified delay.

ixjf, 2019-09-02:
Weird. Pretty sure this is something I tried back when I first looked into this and it worked :/

Arcrocus, 2019-12-11:
I have the same problem on Windows boot on my MSI X470.
And that's how I solved it.
I opened a program that is called CCleaner Professional.
Then TOOLS - STARTUP - WINDOWS SERVICES - and turned off "MysticLight2_Service", save changes and reboot.
After that MSIRGB.exe is running correctly and didn't turn effects to default after every boot,

Martinocom, 2019-12-11:
I found (maybe) another workaround: I just set the service as "Delayed Start".

1) In Windows 10 search for "Services". 
2) Open it and find "MSIRGB.ScriptService". 
3) Left click, properties.
4) Set launch options  as "Delayed Start" and not just "Auto"

(italian example image below)
[image]

Martinocom, 2019-12-12:
> I have the same problem on Windows boot on my MSI X470.
> And that's how I solved it.
> I opened a program that is called CCleaner Professional.
> Then TOOLS - STARTUP - WINDOWS SERVICES - and turned off "MysticLight2_Service", saved changes and reboot.
> After that MSIRGB.exe is running correctly and didn't turn effects to default after every boot,

The problem with my Z270 m7 is that I haven't that service on my PC and it's still not working. I tried to delay the MSIRGB service, but in the end it failed to work, as usual.

ixjf, 2019-12-13:
Indeed, stopping Mystic Light does not solve the problem. I have been able to reproduce the issue on my machine and I don't have Mystic Light installed. The problem is really with MSIRGB.

Unfortunately, I still can't work on this.

Disims3, 2020-06-18:
I'm interested in looking into a work around for applying effects on boot and wondering if the pop  warning about me that  my "b450 tomahawk max board is not supported" is block any attempt. 

 Any chance  you could add it as a supported device with a disclaimer about the colour issue? or  send me a copy with it supported?

ixjf, 2020-06-18:
I can't, unfortunately. I do not have access to my desktop computer where I am right now. I don't have a Windows machine either, let alone the dev tools to build MSIRGB. Maybe in a month or so I'll be able to provide some updates.

jacklul, 2020-06-22:
About the bug after suspend and then resuming to find the scripts are no longer working:
Probably this could be implemented into the service executable - https://docs.microsoft.com/en-us/dotnet/api/microsoft.win32.systemevents.powermodechanged?redirectedfrom=MSDN&view=dotnet-plat-ext-3.1

The workaround with task scheduler is working but is not really a clean solution, it should be handled by the service itself.

ixjf, 2021-08-23:
> I'm interested in looking into a work around for applying effects on boot and thinking that the pop-up warning me that my "b450 tomahawk max board is not supported" is wall.
> 
> Any chance you could add it as a supported device with a disclaimer about the colour issue? or send me a copy with it supported?

What's this about B450 TOMAHAWK MAX not being supported?

ixjf, 2021-08-23:
Can you guys try the updated Scripts? (You have to download the Scripts folder from the source individually, since I haven't released an update to MSIRGB). Also see #159 (comment with the code example) to see how you can apply it to your own code.

closed 2021-09-02 by ixjf