ixjf/MSIRGB #45: Possible to MQTT the color?

t0bse, 2019-06-12:
Hi,
first: nice work, very smart tool, i love it! :-)

I got an LED Light Stripe with Wif Controller integrated to Home-Assistant.
Would it be possible to send the current color to an MQTT Server?

The reason behind this is, that i want to sync my Light Stripe with the Mystic Lights. MQTT would make this possible. Another way would be the other direction, a API to receive a color for the LEDs.

ixjf, 2019-06-22:
Sorry, I didn't reply immediately because I didn't know what to do of this. I would have to do some research in order to implement this, and right now I don't have the time for it. If you or someone else wants to send a pull request, I'll look into it.

ixjf, 2021-02-04:
You'll be able to do this now with the (soon to be published) 2.4.0 release of MSIRGB. Lua scripts can now use any Lua function, which means you can load 3rd party Lua libraries like *luamqtt*.

closed 2021-02-04 by ixjf