ixjf/MSIRGB #52: Fails to open on Windows 7

PrismaStratum, 2019-08-02:
I know you may be unlikely to make the program compatible with Windows 7, but I would really like to use it on my system, so I am going to request it. Currently it will tell me when conflicting software are installed, however, after all conflicting software has been removed and I try to open MSIRGB, nothing at all happens.

ixjf, 2019-08-27:
I'm sorry for the delay. I can look into it, but I can't work on MSIRGB for the moment (or any time soon). I'll look into this when I can. Can you test older versions of MSIRGB and tell me if any of them work?

PrismaStratum, 2019-09-09:
I have tried every major version all the way to v1.0.0 and "Initial release" and the problem remains the same. The process shows up in task manager for half a second then disappears. It doesn't idle in the background or anything, the process terminates instantly.

ixjf, 2020-09-05:
Do you still need this?

closed 2021-02-04 by ixjf