ixjf/MSIRGB #53: msi b450m pro vdh plus is working

ciriys, 2019-08-19:
i dont know how to report this but i have installed and ran it and it works, even with a random cheapo rgb strip

ixjf, 2019-08-27:
Thanks! Will add to the list.

ixjf, 2019-08-27:
Do you get a warning about unsupported model when you start MSIRGB? I'm trying to figure out which revision your motherboard is, since from my knowledge revisions 1.x and 2.x shouldn't work.

ciriys, 2019-08-29:
> Do you get a warning about unsupported model when you start MSIRGB? I'm trying to figure out which revision your motherboard is, since from my knowledge revisions 1.x and 2.x shouldn't work.

yes, i do get the warning at the beginning of app startup but as far as i can tell everything works fine the colors are accurate and even the pre loaded lua scripts work as intended
i am not exactly sure what revision my motherboard is, if you're willing to tell me how to find it, i will try my best to find it and respond to you with the revision, i will attach a cpuid screenshot of the motherboard, hope that helps
also thank you for making such an awesome software with a nice gui for dumb-dumbs like me :)
[image]

ixjf, 2019-08-31:
I'm assuming "7.0" is the version number, but run this in a command prompt to make sure:

`wmic path Win32_BaseBoard get Version`

If it is indeed 7.0, then this is a newer revision and I'll have to add it to the list.

ciriys, 2019-08-31:
yup, you were right it is version 7.0 
[image]

ixjf, 2019-09-01:
Alright, thanks!

commit 7d2ff840 (2020-09-05): Fix #53
closed 2020-09-05 by ixjf