ixjf/MSIRGB #55: Script does not run after resuming from sleep

mrcaseyrose, 2019-09-21:
My script runs just fine when opening the app, or during the initial boot of the computer; however, if I put the computer to sleep and resume again, it does not run and the LEDs return to their default (all white) state.

ixjf, 2019-09-22:
This may be the same as #43. I'll look into it when I can (but can't promise when).

mrcaseyrose, 2019-09-23:
Hey thanks for taking the time to reply. Does it work fine for others upon resuming standby? I was thinking it was a limitation of the RGB controller in the motherboard.

ixjf, 2019-09-28:
I don't honestly know what it is, but it's probably something that I can handle from MSIRGB, _if_ it's not a bug.

Cris--, 2019-12-16:
[Here my solution](https://github.com/ixjf/MSIRGB/issues/69#issuecomment-566262559) with Windows Task Scheduler..

ixjf, 2020-09-06:
Closing as it is a duplicate of #43. Please refer to that issue for info on the status of the fix.

closed 2020-09-06 by ixjf