ixjf/MSIRGB #56: Works on MSI H270 Gaming Pro Carbon

HighwayChile, 2019-09-25:
Thank you so much! MSI Gaming App blows!

ixjf, 2019-09-28:
Alright, can you describe the problem a bit further? What do you mean by trouble displaying the colour green?

HighwayChile, 2019-09-28:
Yes, it cycles incredibly fast through a large range of colors when I try to display green. It flashes through RGB at high speed. It flashes one color, then another color, then another color, so fast that it is dazzling. It happens when I set the color to green (0f0) and have the step duration set to 256 and not running scripts from the menu. Breathing effect seems to make no difference with the bug. (I have not tried debugging yet.) 

I haven't played with the program much more. I just set it to a theme I liked and have been watching my voltages. Everything looks normal so far as that goes.

ixjf, 2019-10-04:
In that case, the only thing I can do is do some research on your motherboard. Maybe it works slightly differently. But I can't promise to be able to work on this any time soon (or MSIRGB in general).

HighwayChile, 2019-10-07:
Hey, it's all good. I have a job too, hehe. I was just letting other people
know. I actually tried to replicate the flashing lights issue once and
wasn't able to. I didn't try all the settings, but, it may have been a
minor bug after running one of the scripts. Anyway, I haven't seen any
problems since, so I am using your software! It works great and it has more
options/fewer restrictions than MSI's stupid "gaming app," which seizes
control of some audio options, making Starcraft 2 only playable without
audio, lol. Thanks again!

ixjf, 2019-10-10:
Alrigut, that's good. I'll still try to look into this when I can, though.

commit 7c1182ff (2019-09-28): Fix #56
closed 2020-09-05 by ixjf