ixjf/MSIRGB #58: X370 Gaming Pro Carbon Wrong Lights

MaxenceDeshaies, 2019-10-01:
The X370 Gaming Pro Carbon Wrong Lights has two type of lights;

1. The RGBs
2. Function lights, which are only in red and lightup when there is a connected RAM module, PCIe card, etc.

But the software interracts with the function lights rather then the RGBs...

ixjf, 2019-10-04:
The motherboard probably uses a different chip for controlling the RGBs, so unfortunately currently won't be supported. But it's worth a look when I can (and I can't promise it will be any time soon, sorry).

MaxenceDeshaies, 2019-10-04:
Alright, let me know if you need something, I’ve briefly looked at the code but it’s way to complex for my student knowledge. 🤣

ixjf, 2020-09-05:
At this moment, this motherboard won't be supported.

closed 2020-09-05 by ixjf