ixjf/MSIRGB #60: Not openning in B350 Tomahawk

mamad0901-sys, 2019-10-16:
when I accept the risks , nothing happens . I checked task manager and windows problem reporter was active .

Windows 10 Build 18362
B350 tomahawk with rgb strip in header

I also should note that I use msi gaming app to control rgb header becuase mystic light doesn't work in this motherboard. I deleted gaming app but it solved nothing .

mamad0901-sys, 2019-10-16:
Solved by closing msi afterburner 
its working thanks!

closed 2019-10-16 by mamad0901-sys

ixjf, 2019-10-21:
Does it always fail to work with MSI Afterburner working?

mamad0991, 2019-11-14:
Yes . I should always close it .

ixjf, 2019-11-16:
Alright. That's good. I'll keep this in mind and see if there's any way to work around it without changing that setting.