ixjf/MSIRGB #61: Not working on MSI X470 GAMING M7 AC

mdogmano, 2019-11-07:
Lights just stay off. Did I miss something?

I uninstalled Mystic Light after switching lights to off. Neither manually applying colors or running scripts doesn't work.

ixjf, 2019-11-16:
Can you run this on a command line and show me the result?

`wmic path Win32_BaseBoard get Product, Version`

It should tell me the exact motherboard revision. I don't know why it doesn't work, but that will help.

Anineta, 2020-01-26:
C:\Users\proDu>wmic path Win32_BaseBoard get Product, Version
Product                      Version
X470 GAMING M7 AC (MS-7B77)  1.0

ixjf, 2020-01-30:
Okay, weird. I'll look into this when I can, but I can't promise that will happen any time soon :(

If you turn on the lights with Mystic Light, does MSIRGB work after?

Anineta, 2020-01-31:
Sadly, no.
MSIRGB doesn't affect to RGB lightning at all,. I've tested differenc scenarios, but have zero luck. There's no difference if it was launched with Mystic Light off/on, or after deinstalling Mystic Light and rebooting with RGB turned off/on.

ixjf, 2020-01-31:
Okay, I will look into this when I can. Thanks for reporting.

ixjf, 2020-09-05:
Unfortunately, this motherboard is not 100% supported by MSIRGB at this moment. I might look into this at some point in the future.

closed 2020-09-05 by ixjf