ixjf/MSIRGB #69: MSI Z370 PC Pro (7B49-001R) works but...

Cris--, 2019-12-05:
Hi,

thanks for this great Mystic Light alternative!
why i'm using MSIRGB: I block Mystic Light with the Windows Firewall (all from ML what i can find) and after a reboot the firewall rules for the ML gets reset without denied rules anymore!!! Absolutely a no go!!! 

What i'm missing is a proper autostart solution for the MSIRGB. After a cold reboot the LEDs are white. The MSI Z370 PC Pro have no EEPROM to store the LED conf. So i need something to autostart a MSIRGB lua script (task scheduler, autorun from registry or startup folder).

Is there something already or planned?

ixjf, 2019-12-10:
This is a known issue. I won't be able to solve this any time soon :(

closed 2019-12-10 by ixjf

Cris--, 2019-12-16:
Ok, i managed it with the Windows Task Scheduler. It's simply restart the service:

Run with highest privileges
Trigger: Login, System Start, unlock Desktop
Run Program: cmd
Arguments: /c "net stop "MSIRGB.ScriptService" & sc start "MSIRGB.ScriptService""

ixjf, 2019-12-17:
That is nothing more than simply rerunning MSIRGB. It is not a real solution to the problem. I still have to look into this, but if that works for you, that's good.

Cris--, 2019-12-17:
Thats right but i'm not a coder. So for me it solves the start issue for me!

Cris--, 2019-12-17:
Also many apps like MSI Afterburner, AeroGlass, Samsung Magician or MS Hardware ipoint/itype are starts from the task scheduler after install. Why not MSIRGB with a rerunning?  ^^

ixjf, 2019-12-18:
MSIRGB does start automatically on Windows startup if a script was set to run previously. For settings applied directly through the GUI, it doesn't, and it's currently not feasible to do (despite me wanting to) either due to the way I designed the code.