ixjf/MSIRGB #71: Any command line arguments supported?

przekrencone, 2019-12-21:
For example calling "MSIRGB.exe f f f" turns LED white.

ixjf, 2019-12-21:
No, there aren't. What would be the use case?

Cris--, 2019-12-21:
Start minimized with e.g. Windows Autoruns with a specific color and maybe brightness. ^^
MSIRGB.exe #ffffff b50

ixjf, 2019-12-23:
You can already do this with a script. As for brightness, it's not supported by the hardware as far as I know.

przekrencone, 2019-12-23:
You can of course change brightness by changing colors down. F->8
On my motherboard (b350m mortar) red channel works very smooth, but blue always stays at 100%

Cris--, 2019-12-23:
Can anyone share a script with one color only and no fade please?

przekrencone, 2019-12-23:
Give me 10minuts, i have it on pc

przekrencone, 2019-12-23:
> Can anyone share a script with one color only and no fade please?
Here you go
[Solid Green.zip](https://github.com/ixjf/MSIRGB/files/3995096/Solid.Green.zip)

closed 2020-09-05 by ixjf