ixjf/MSIRGB #75: Not Working on MSI Z370 Gaming Pro Carbon

engine20, 2019-12-29:
I downloaded the tool, set the LEDs in Mystic light to off, restarted and launched MSIRGB. But there ist nothing happening. I tried installing Mystic light and uninstalled it again after turning the LEDS on, but no effect. :(
Can anybody help me?

ixjf, 2019-12-29:
Did you disable Mystic Light? Setting the LEDs to off in Mystic Light doesn't stop the program.

engine20, 2019-12-29:
I uninstalled it too

ixjf, 2019-12-29:
Make sure the service "MysticLight_Service" is stopped before running MSIRGB.

engine20, 2019-12-29:
Ok, it isnt running anymore, now i can see it working on a few tiny LEDs on the Mainboard, like the LEDs at the RAM banks. Am I abled to change this?

ixjf, 2019-12-29:
The chip that controls the LEDs usually changes the color of all LEDs. I'm not aware of any way to change only specific LEDs.

engine20, 2019-12-29:
The Problem is that it affects only one Group of LEDs.

ixjf, 2019-12-29:
In that case, MSIRGB probably isn't compatible with the motherbord. I'll have to look into it later.

engine20, 2019-12-29:
Ok. :/. But i still appreciate the fact that there are people out there doing stuff like that for free.

Opisek, 2020-05-13:
I can report that it doesn't work ony my Z370 Gaming Pro Carbon either. Uninstalled Mystic Light, rebooted the pc. The buttons do not seem to be having an effect and the default rainbow effect continues to be played.

ixjf, 2020-05-15:
I may have overlooked the fact that it doesn't control all LEDs on some motherboards. Unfortunately I can't do anything about this right now as I don't have access to my desktop computer.

Opisek, 2020-05-15:
Nevertheless, thank you very much for all your efforts and for creating this piece of software! Would love to see a support for this motherboard as well, if it's not too difficult to implement.

jackhuii, 2020-07-14:
Same issue also applies to MSI MPG Z390 Gaming Edge AC. However I have been trying to find a way to turn off the DRAM LEDs so the fact that I can use this to do so helped me! :)

ixjf, 2020-09-06:
This motherboard will not be supported at this moment, unfortunately. I'm closing this issue for now.

closed 2020-09-06 by ixjf