ixjf/MSIRGB #82: Switch back to MSI Mystic Light possible?

Frechbass, 2020-02-06:
Hi there,
not sure if this is normal. 
Got a "MSI B450 Tomahawk Max" and now, MSI Mystic Light won't open up again after using MSIRGB.
PC is creating a instance of MSIML, as i see in the Task-Manager.
Also, "Pumpkin" is blue for me, so the colors are inverted here, while other B450TomahawkMax-Users say it's working flawlessly.
Am i just too dumb?
Greetings,
Frechbass

ixjf, 2020-02-08:
Seems unlikely that MSIRGB would cause Mystic Light not to work again, but I won't rule it out.

Mystic Light shouldn't be running at the same time as MSIRGB. I'm not sure if that's what you are doing.

Other than that, I'm not sure what I can do. Have you tried reinstalling Mystic Light?

ixjf, 2020-09-05:
Did you fix this?

Frechbass, 2020-11-18:
Sorry for the silence...
Yeah, a reinstall fixed it.
After your update im hooked to your version again, everythings fine now.
No inverted colors, or something like that.

Thought about to do some custom-scripts in the future.
We will see.

Thanks for fixin' & have a nice day. :)

closed 2020-11-18 by Frechbass

ixjf, 2020-11-18:
Great! You too :)