ixjf/MSIRGB #84: X370 Gaming Plus Support

Grimreaper1979a, 2020-02-21:
Any chance of support for the X370 Gaming Pro? I tried it, but it didn't work at all. The standard RGB support from MSI just plain sucks compared to other boards I have from the like of ASUS. And just doesn't work right. I would really love to see support for this board.

ixjf, 2020-03-03:
I can't say. Maybe yes, maybe not. Either way, unfortunately, I can't work on MSIRGB until the summer, so I'll leave this on hold.

ixjf, 2020-09-05:
It's weird that you say it doesn't work because it has been reported to work on the X370 GAMING PLUS. Can you confirm? Is this the PRO or the PLUS?

Grimreaper1979a, 2020-09-05:
It's the plus, and sometimes it works, and sometimes it doesn't. Meaning
there are times it just doesn't do anything at all, and after either
rebooting, or killing the program in Task Manager and relaunching it it
works. Also there are quite a bit of issues with color matching, even
between fans of the exact same model, that when used in my ASUS machine,
the colors all match perfectly, but on the MSI, they are off by a few color
tones.

ixjf, 2020-09-05:
Can you try the new version of MSIRGB and report back? It's v2.3.0.

ixjf, 2020-09-06:
Closing as it is a duplicate of #139. Please provide any input on that issue. Make sure to state there what motherboard you're using.

closed 2020-09-06 by ixjf