ixjf/MSIRGB #89: B450-A PRO MAX working

Shihrt, 2020-03-21:
Motherboard not bricked, colors working fine.

[image]

oncle-anatole, 2020-04-13:
Are you sure it works correctly because I've the same exact motherboard with the same bios but the behaviour seems a bit random. Unless I've not understood how the software works. It seems when clicking on buttons, the software doesn't register half of the time. Also, the colors are wrong. For instance I put blue on all the squares and the effect I've is pulsing between white (default) and purple. Whatever combination of colors I try to enter, it never shows red. Enabling breathing makes my strips pulse from off to white, no color. etc. Mystic Light isn't installed by the way but I guess I unfortunately will need to because this software doesn't seem to work like intended, it's just too erratic. The scripts seem to work though

ixjf, 2020-09-05:
> Are you sure it works correctly because I've the same exact motherboard with the same bios but the behaviour seems a bit random. Unless I've not understood how the software works. It seems when [...]

Guaranteed you have some program interfering with your LEDs. That's the standard behaviour.

ixjf, 2020-09-05:
> Motherboard not bricked, colors working fine.
> 
> [image]

Thanks!

closed 2020-09-05 by ixjf