ixjf/MSIRGB #90: b450 Tomahawk MAX weird coloring patterns

FjellsMemory, 2020-03-25:
Hi there, not only do all colors appear reversed for me, but between each color change, regardless if using script or manually selected colors, the color white will appear one time, before moving onto the next color in the series.

ixjf, 2020-03-26:
Make sure Mystic Light is not running, even the Mystic Light _**service**_.

FjellsMemory, 2020-03-27:
thx for reply.  i don't think it is.  certainly no .exe running named MSI or Mystic anything is running.  however, i have gskill ram, amd wraith prism cooler, and powercolor rx 5700xt, all of which have proprietary software installed to control their lighting.  conceivable one of them might be interfering, although i have none of them actively running while trying to use your app.  still puzzled about this issue.

ixjf, 2020-03-27:
Did you check if the service was running? What happens if you stop MSIRGB? Do the RGB turn white, or do they stay the color they were supposed to be with MSIRGB?

padrim, 2020-04-03:
I'm having this issue as well, oddly, with the same motherboard. I came here to check if anyone had also noticed it. It only started recently, after having the PC off for a couple of days. I don't have Mystic Light installed currently, as I am using this one. I do have some other control software though, iCUE for a mouse and 2 different g.skill software, on for RAM, one for a keyboard. I tried killing all of those and it didn't do anything. All the settings will do this, fade between whatever color it should be and white, even the scripts. If I turn on breathing mode it only breathes in white. Of course, all the colors are input backward as well. (0=f, f=0) If I disable all lighting from within the app the lights all go off. I don't see any process running that has to do with this app so I can't see any other way to stop it.

FjellsMemory, 2020-04-03:
@padrim your experiences, current software, and descriptions are exactly mine all the way across the board.

padrim, 2020-04-03:
It was working fine except for the already reported reverse input for some time. This is recent. @ixjf can you recommend how to reset the software or hardware? I don't see any processes running while the app is open or closed, and the last settings remain even after restarting the PC. I guess next I'll try to reinstall Mystic Light and see if that will override the settings in the hardware now. I don't think there is anything specific about this motherboard that makes it different from any of the others that use the same hardware, but I don't know. I thought maybe a Windows update. I'll install Mystic Light and report back.

FjellsMemory, 2020-04-03:
> I don't see any processes running while the app is open or closed, and the last settings remain even after restarting the PC. I guess next I'll try to reinstall Mystic Light and see if that will [...]

All of this mirrors my own experience 100%.  I really just want to be able to assign a single static color.  That's my only desire.  But I can't via this software atm.  There is no way to stop colors changing from whatever to white and then whatever else and then white endlessly.  Even the scripts do it.

padrim, 2020-04-03:
@FjellsMemory why not just use Mystic Light?

padrim, 2020-04-03:
So I installed MSI Dragon Center, as the MSI web page for the Tomahawk B450 Max listed it as having been updated last week to work with the RGB and no longer lists Mystic Light as a download on the page. It was buggy and didn't work well, and didn't sync the other RGB products properly. It also made MSIRGB not work properly at all. I uninstalled it and now MSIRGB is working as it was, with the colors reversed but otherwise properly, including the scripts. I assume installing Mystic Light again would have the same effect. The G.skill software for the RAM has stopped working altogether though. I may have to uninstall and reinstall that too. I wish all the RGB hardware companies would come up with something that works together.

FjellsMemory, 2020-04-03:
@padrim - because one of the .exe used in mystic light is flagged as a cheat in two of my online games!  XD  (Vermintide 2 and Apex Legends)  that's how i began looking for a lighting alternative in the first place.  by chance i also use g.skill ram and find your experiment with dragon interesting, as when i built this rig just 3 weeks ago, dragon center was only for other boards.  thanks for your updates.  i will probably try mystic light and or dragon center myself, given that at least you got MSIRGB to work, albeit reversed.  that would be fine with me, as long as this intermittent white thing would stop.  but perhaps one of our common peripheral lighting softwares (like the g.skill) somehow is the source of this issue.  hmmm... like you, i wish the various companies would come to some common solutions.

FjellsMemory, 2020-04-04:
welp, just thought i'd touch base again and report that i, too, installed msi dragon center.  this move changed the white intermittent lighting problem i was having with MSIRGB.  the problem went away, but, like padrim, the color coding remains reversed to what the user sets in MSIRGB.  that is not a problem for me.  i can get what i was after:  setting a single static color.  interestingly, tho, i am also able to use dragon center without issue in my games.  this was not the case when using mystic light on its own.  so.  thanks both for your time in this thread.  any others having problems, i recommend you turn to dragon center and try your luck.  tho with dragon center one has full light control already, so MSIRGB becomes sort of moot at that point, if you're not using it for the scripts.  peace!

ixjf, 2020-04-04:
MSI Dragon Center probably overrides some setting that was wrong previously in the chip (I'm guessing). If that is the case, then there are other ways that don't involve installing any other software (e.g. turning the PC off, turning the power supply off, pressing the power button for about 10 seconds, and turning everything back off, or, even, removing the CMOS battery for about 10 minutes).

Regarding the color reversing, that is a known issue. It should have been fixed already but I couldn't yet since I don't have access to my computer until the summer.

Jacobite747, 2020-05-13:
I have also similar issues on B450 Tomahawk. I noticed using the Pumpkin script as an inital test bench as well as the other scripts, that there appears to be some sort of default "breathing" activity that is present in all lighting configurations such as the Hue Wheel even when breathing is disabled in the script or GUI. This might be something disabled when MSI DC is installed as per @FjellsMemory experince. 

With this in mind, plus the fact that the colours are reversed on this board, the default breathing produces this intermittent white colour, which is probably _supposed_ to be black.

Unfortunately I don't know where that default breathing would come from...

ixjf, 2020-09-05:
MSIRGB v2.3.0 fixes the reversed colour issue. Please try the new version.

ixjf, 2020-09-06:
Closing as it is a duplicate of #139.

closed 2020-09-06 by ixjf