ixjf/MSIRGB #91: B350 gaming plus does not work

Schrema1, 2020-03-29:
Hello,
I've tried using the Software and everithing was working fine, untill i hit apply. since then the rgb headder is dead. is there any way to remove/ undo the program?

Thank you in Advance.

ixjf, 2020-03-30:
You can either:

- run MSI Mystic Light;
- turn your PC off (power supply button off), press the power button for several seconds (>10 seconds), and turn your PC on again. that should reset the chip to normal.

When you do this, please try rerunning MSIRGB and tell me if it works, because this motherboard had previously been reported to work.

Schrema1, 2020-03-31:
hey, i guess the problem lies with my mobo, the msi software is allready really unstable, and it took me a lot of trial and error to get it working again.
So I guess your software should work again as well, but i don't want to redo everything I did to get it back to working again if it does not work.

closed 2020-03-31 by Schrema1