ixjf/MSIRGB #95: Works on X370 Gaming plus, temperature script and question

btd5, 2020-04-10:
-Works on X370 Gaming plus.
-Have added a little script, that controls the color from temperature of cpu and gpu with backup color wheel in case aida isn't running.
-Could you add "rainbow" HSV  like Fast LED uses or help me add it myself. "Spectrum" HSV has little to no yellow.
Thanks.

[Temperature wheel.txt](https://github.com/ixjf/MSIRGB/files/4461836/Temperature.wheel.txt)

ixjf, 2020-09-05:
Thanks for the input. Will look into the script when I can.

ixjf, 2021-02-04:
Can you explain your reasoning? Why that formula for T? Why `(1.666 - T)%1`? (which is equivalent to `1.666 - T`, by the way.

Also, could you open a pull request instead?

I must also apologise for how long I took. I can't remember why it took me so long, but I know April-August was around quarantine time and I wasn't having it very easy.

btd5, 2021-08-21:
Sorry for no response on my side, I have switched motherboard in the meanwhile and now have a Gigabyte (therefor incompatible with the code). `(1.666 - T)%1` is because the value had to be between 0 and 1 (0,666 is blue, 0,333 is green, 0 and 1 is red) and I wanted it to go from blue as cold to yellow and then red when hot.

ixjf, 2021-08-22:
Alright. About the rainbow effect, I don't know if this isn't a hardware limitation? Hue Wheel should already go through yellow equally as much as other colours, _I think_.

ixjf, 2021-09-03:
I'll let someone else deal with the rainbow thing.

closed 2021-09-03 by ixjf