ixjf/MSIRGB #96: Any way to get this working with peripherals?

RafeeSamith, 2020-04-13:
I know this is for motherboards and their RGB headers only but I've been trying to set my MSI CLUTCH GM11's RGB to orange forever and Dragon Center only lets me choose between 7 colors. My lack of experience in C++ and Visual Studio isn't helping either since I tried using SimoDax's tool but to no avail. Any ideas? Thanks in advance c:

Mayravixx, 2020-04-21:
Same here, I've been told previously that MSI's Mystic Light software can be used for Razer products since Synapse is now shitting the bed, and somehow I ended up here. That being said though, I would like to see this program working with my GPU if possible, since I do have an MSI RX 580

ixjf, 2020-09-05:
https://gitlab.com/CalcProgrammer1/OpenRGB

Might do what you want.

closed 2020-09-05 by ixjf