ixjf/MSIRGB #98: DriverLoadFailed exception

tomko80, 2020-04-20:
When I run the msirgb exe I get the warning that my MB is not supported but when I ignore that the app starts and the leds work just fine.

I now tried to call the dll directly in my own (C#) command line app using the (almos) original TryInitializeDll function to fire the things up. On the first run I get first the expected MB error but on the second it fails in a DriverLoadFailed exception thrown by the sio part.

Anything special I should know to consider in the project config e.g.? Can't figure out why the original apps works but my doesn't.

ixjf, 2020-04-21:
Make sure you're running your app as administrator.

tomko80, 2020-04-21:
That was it, thanks!

closed 2020-04-21 by tomko80