ixjf/MSIRGB #100: Works with Z390 Gaming Plus but reversed Colours

JorkdaOrk, 2020-05-01:
Works generally with a Z390 Gaming Plus, but colors are reversed. 
After starting a script, colors are reversed in Mystic Light as well.  
To reset colors in Mystic Light, just turn off  PC, pull power cord, hold the off button for 15 seconds. 
Than the colors switch back to normal under Mystic Light.. 

Could create a beatiful Orange Lighting, which seemed to ber impossible under Mystic Light. 

Wrote the script "Basic Color Cycle" for noobs like me who just want to edit some colors. 

[Basic_Color_Cycle.zip](https://github.com/ixjf/MSIRGB/files/4566700/Basic_Color_Cycle.zip)

ixjf, 2020-09-06:
Version 2.3.0 should fix the issue of reversed colours. Please retry. If it still doesn't work, please open a new issue.

As for your script, if you're setting all the colours to the same RGB value, you don't need to change step duration. You also don't need a while loop nor a os.sleep. You only need to set the colours once.

closed 2020-09-06 by ixjf

JorkdaOrk, 2020-09-06:
tested it, and it works fine with my Z390 Gaming Plus. 
THX

ixjf, 2020-09-06:
Great! Thanks!