ixjf/MSIRGB #103: MSI B450 Gaming Pro Carbon AC

pguerrerox, 2020-05-04:
Has anyone get this software working on this motherboard?  i need an alternative to mystic light.

ixjf, 2020-05-23:
Why doesn't it work?

benchebanianas, 2020-06-02:
@pguerrerox i am getting this 

[image]

ixjf, 2020-06-03:
Can you run this command on a command prompt:

`wmic path Win32_BaseBoard get Manufacturer`

and provide me with the result?

MrPalamen, 2020-06-03:
Manufacturer
Micro-Star International Co., Ltd.

aaronthew, 2020-06-05:
Same here on the b450 Gaming Pro Carbon AC  
wmic path Win32_BaseBoard get Manufacturer
Manufacturer
Micro-Star International Co., Ltd.

benchebanianas, 2020-06-05:
@ixjf i found out why i was having this problem: basically i am also running MacOs in my computer and it changes the Motherboard manufacturer name in order to boot. After changing the bios to the Windows 10 one it runs normally

aaronthew, 2020-06-05:
@benchebanians that's for the update, I don't have Macos on my pc but will look into bios issues

ixjf, 2020-06-06:
In the cases where WMI returns the correct manufacturer, I'm not sure why it is not working. Unfortunately, I can't debug this until end of July/start of August since I don't have access to my work desktop :/

XenSide, 2020-08-19:
I can confirm that this board is incompatible. 
I can run the program just fine without errors but applying colors or scripts does nothing at all

ixjf, 2020-09-05:
Indeed, it is.

closed 2020-09-05 by ixjf