ixjf/MSIRGB #104: Problem load driver

cassiokennedy, 2020-05-05:
Can someone help me? Failed to load the b450 network card driver and I was able to use the program only once, when the new opening appears in the following message "Failed to load the driver. This may be due to some program interfering with the MSIRGB driver"

ixjf, 2020-05-05:
It most likely means you have some other program that is using _inpoutx64_ as well, the driver MSIRGB uses to control the hardware. Do you have any software installed for controlling some hardware, other than MSI's?

cassiokennedy, 2020-05-05:
I uninstalled everything and the software opened, but I choose the color for example red and it is blue. Could it be incompatibility?

ixjf, 2020-05-07:
Which motherboard do you have?

cassiokennedy, 2020-05-07:
MSI b450 Tomahawk

ixjf, 2020-05-08:
It has been previously reported that for some motherboards **(including yours)**, the colour codes are inverted. Unfortunately I can't work on MSIRGB at the moment, otherwise I would have pushed a fix already.

cassiokennedy, 2020-05-08:
No problems and congratulations for the project

velasd4, 2020-05-12:
If you have Riot Vanguard installed, turn it off then try opening MSIRGB again.

closed 2020-05-15 by ixjf