ixjf/MSIRGB #105: B250M PRO VDH, 7A70vA8, failed to load driver error

Pawulon404, 2020-05-05:
I have this error when I want to run MSI RGB.exe 
I don't have mysticlight installed but I have fast boot and afterburner. I also have razer synapse for mouse settings. 
This motherboard is incompatible or should I delete other msi applications too?

ixjf, 2020-05-05:
It might be incompatible but you wouldn't get that error if that were the case. Most likely, you have some program running which uses the same driver that MSIRGB uses (inpoutx64). Please try stopping all software (and associated services) that might be accessing the hardware (including the ones you mentioned). Then start one by and one and see which one causes the issue.

Pawulon404, 2020-05-08:
I installed and uninstalled mysticlight again, now msirgb works. Razer synapse did not cause a problem

closed 2020-05-08 by ixjf