ixjf/MSIRGB #107: Driver issue - B450 Tomahawk Max

krov493, 2020-05-08:
When attempting to run MSIRGB.exe, I get a Failed to load Driver message. Brand new install of Windows, and have removed MSI Dragon as well as any other controller software. 

I noticed I don't have inpoutx64 in System32, and when I run `sc query MSIRGB.Driver` I get a return of "The specified service does not exist as an installed service".

Any suggestions on other things to try?

ixjf, 2020-05-09:
`sc query inpoutx64` would be the correct command now, as I changed the name that MSIRGB registers the driver as.

adbeem, 2020-06-01:
Hi, may i know if there is a fix for this? I have the same motherboard and same error "Failed to load driver..."
I also ran "sc query inpoutx64" and got this message:

[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.

ixjf, 2020-06-02:
This is weird. Is it a brand new installation of Windows as well?

Could both of you run this command on a command prompt and upload here the file:

`tasklist /svc > output.txt`

This will print a list of all running processes and services to a file 'output.txt' that will be saved in the directory where the command prompt is running.

PAlbrechtHH, 2020-06-08:
I am having the same issues.
Had it all running on my B350 Gaming Pro Carbon but now I am getting "Failed to load driver"  when launching MSIRGB.exe.
[image]

Executing sc query inpoutx64 also gives me 
`[SC] EnumQueryServicesStatus:OpenService FAILED 1060:

The specified service does not exist as an installed service.`

The only thing I changed about my system is updating the AMD chipset driver to the newest version (2.04.28.626).

Maybe there is an compatibility issue when using the newest chipset driver?

ixjf, 2020-06-08:
If you could try to rollback, that would be good. Please also see [this](https://github.com/ixjf/MSIRGB/issues/107#issuecomment-637440864) issue.

PAlbrechtHH, 2020-06-08:
Unfortunately I am not able to revert to the old chipset driver. The old one simply does not install anymore since I installed the newer one.

Attached please find output.txt file
[output.txt](https://github.com/ixjf/MSIRGB/files/4747149/output.txt)

ixjf, 2020-06-08:
I would recommend that you start _msconfig_ and select "Diagnostic startup", "Apply" and reboot. Then try running MSIRGB. If MSIRGB works, then enable non-Microsoft services one by one ("Services" tab), rebooting each time to find which program is causing incompatibility. Once you find it, you may go back to _msconfig_ and select "Normal startup" again, making sure that the incompatible program is not enabled.

PAlbrechtHH, 2020-06-10:
> I would recommend that you start _msconfig_ and select "Diagnostic startup", "Apply" and reboot. Then try running MSIRGB. If MSIRGB works, then enable non-Microsoft services one by one ("Services" [...]

Thanks for your reply!
Unfortunately the software is not working in diagnostic startup mode neither.
When attempting to run MSIRGB.exe in diagnostic startup mode nothing happens.
The error message popup mentioned above does not get shown in this mode.
Is there something else I can try?
Is the software even compatible with newest Windows 10 release and chipset drivers?

ixjf, 2020-06-11:
Sorry, it's likely that Diagnostic Startup disables services that are required for MSIRGB to work. Instead, do Normal startup, but go to "Services", tick "Hide all Microsoft services", and disable all. If any startup item that isn't from Microsoft is enabled, disable it (Task Manager -> Startup).

ixjf, 2020-09-05:
Did you fix the issue?

PAlbrechtHH, 2020-09-05:
Yes I fixed it by uninstalling Valorant.

ixjf, 2020-09-05:
Alright, thanks!

closed 2020-09-05 by ixjf