ixjf/MSIRGB #108: MSI X470 GAMING PRO CARBON - Failed to load driver.

Untalia, 2020-05-10:
I thought the motherboard was on the -working- list.
Maybe some software is interfering with this program.
Is there a way to check it /  a log?
 
I made sure to disable iCue, Msi Afterburner and also uninstalled latest edition of Mystic Light 3

ixjf, 2020-05-11:
This error is not due to some incompatibility with the motherboard. It's usually due to some other program using the same driver that MSIRGB uses to acess the hardware. 

It's hard to tell without access to my desktop computer what programs might be causing this (this is a common issue recently, for some reason), but someone reported that uninstalling and reinstalling Mystic Light solved the problem for them. If that doesn't work and it isn't too inconvenient, I'd ask you to _uninstall_ any program that controls hardware and try again.

Untalia, 2020-05-11:
After each uninstall I restarted and tried again :
Corsair iCue
MSI Afterburner
Installed and Uninstalled Mystic Light
Riot Vanguard

Riot Vanguard fixed the issue of the program not running but now I am having a different problem. The only LEDs I can control are the two above the ram slots ( red ones ) and one led from the Graphics Card (rtx2060 msi).
Any ideas?

ixjf, 2020-05-15:
It might be that the motherboard is not actually 100% supported. Maybe some of the LEDs are controlled in a different manner. In which case, there's not much I can do, since I can't write a program for a motherboard I don't own and isn't similar to mine. Sorry.

Untalia, 2020-05-16:
I'll format my PC soon, and before I install everything else I'll test it again and see.
If I could do anything in my part please say so, I am studying Computer Science so I can figure stuff out probably, if you guide me and have the time and willingness to do so obviously.
Thanks in advance.

pajicadvance, 2020-05-24:
I can confirm that Riot Vanguard prevents MSIRGB from loading properly. I received a notification from Vanguard that it has been disabled, and after turning off Vanguard the app worked just fine.

ixjf, 2020-09-05:
I was doing some research and as of this moment I can't confirm MSIRGB will work 100% with this motherboard, therefore, I'll remove it from the list of motherboards that MSIRGB should work with.

closed 2020-09-05 by ixjf