ixjf/MSIRGB #110: B350 PC Mate Unable to disable all lighting & colors confused

BruebachL, 2020-05-14:
Heya,
so the tool does work on my B350 PC Mate, complete with scripts, breathing, flashing and everything but the colors seem to be weird.
For one, I can not turn them off completely. My mainboard LED turns off but my RGB fan (The one connected to the actual header. A cooltek 120 Silent RGB.) only dims down. Strangely enough, it dims down to a pure red color. 
Furthermore, all colors seem to kind of have a red touch to them. Setting it to black gives red, setting it to blue gives purple, etc.

BruebachL, 2020-05-15:
The board is running a Nuvoton NCT6795D-M  (ISA A20h) super I/O chip which might handle things slightly differently. I'd love to contribute to this project if you could point me in the right direction on how to research my chip, otherwise I shall investigate on my own.

BruebachL, 2020-05-15:
[image]

Here's a screenshot of Super I/O when set to 0,0,0. Step duration: 511 Breathing mode: Disabled Flashing Speed: Disabled

BruebachL, 2020-05-15:
Actually, that one is all disabled. The only difference between that and applied is F0 in E4 tho.

BruebachL, 2020-05-15:
Interestingly enough, if I connect an RGB LED strip, that one turns off completely, while the rgb fan (connected directly to the header) keeps cycling between a low red dim and a slightly less bright red dim (if cycling between two colors was previously selected in the program.)

JMDL144, 2020-09-05:
Hi, you ever solved this? 

> Interestingly enough, if I connect an RGB LED strip, that one turns off completely, while the rgb fan (connected directly to the header) keeps cycling between a low red dim and a slightly less bright red dim (if cycling between two colors was previously selected in the program.)

BruebachL, 2020-09-05:
At this point, I believe the fan itself might just be unable to shut down completely, since the LED strip receives the correct signal. I had another issue with a bad PCI-E riser cable wrecking havoc on the NVIDIA driver so that took up most of my attention lately. You can consider this issue resolved. Program works as expected on a B350 PC Mate.

closed 2020-09-05 by BruebachL