ixjf/MSIRGB #113: Changing colours has no effect on b450 tomahawk max

FeSong, 2020-06-01:
Hi there,

Just typing this up as your app is awesome!
Minus one flaw, I can't get the solid colour or any to change anymore. All scripts work and the disable LED works as well but I believe after the first time I played with changing the LED colour, it is stuck on blue. 
[image]
I've tried to set it as that to get that colour set as solid on the mobo but it wont change. I can run Hue wheel and it runs fine with changing colours just cant change the one colour. 
In addition, the hue wheel never really allowed me to set those colours. It just goes all rgb.

ixjf, 2020-06-02:
Does this continue after restarting the computer? Can you shut down your computer, turn the power supply off, press the power button for about 10 seconds, and then turn everything back on? It should reset the chip. I'm not sure what the problem is since the script uses the same mechanism to change colours as the GUI, but maybe it will work.

FeSong, 2020-06-02:
Hey thanks for the quick response, I tried that as you said and it did indeed reset it but its made no real difference. I tried this:
[image]

It just gives me a solid white/bluish light now :/ 
If I play around with the hue wheel a lot:
Let it change colours for about 5 secs then stop it and pray I get the colour I want and it'll give me a different solid colour when I remove hue wheel. Not sure whats happening lol

ixjf, 2020-06-03:
But the scripts apply the correct colors?

panicgale, 2020-06-07:
i can confirm that . it not work well on B450 tomahawk max
the scripts not work too 
the only thing work is DISABLE ALL LIGHTING 
after i delete all file . it does not give me red light back too

FeSong, 2020-06-11:
> But the scripts apply the correct colors?

The hue wheel script only cycles rgb. Selection of colours has no effect on it so I guess not?
Strobe and police lights do work. The pumpkin one I am using at the moment; with lots of fiddling.

I have to use hue wheel to cycle colours, after this, pressing apply stops it on a random colour. I use this with pumpkin to sort of set my systems and wallpapers rgb.

As the other user mentioned, the disable all lighting does work.

ixjf, 2020-08-19:
This is weird as other users have reported that it works with this motherboard. I'll look into it ASAP.

FeSong, 2020-08-29:
> This is weird as other users have reported that it works with this motherboard. I'll look into it ASAP.

Thanks, look forward to the fix as this is a fantastic piece of software, just wish it would work with my mobo lol. Puts MSI to shame with the rubbish they have released to control the rgb. All I want to do is to be able to pick my colours even if they are solid lol.

Cheers mate

ixjf, 2020-09-05:
Did you fix this? Try the new version (v2.3.0) of MSIRGB.

FeSong, 2020-09-05:
> Did you fix this? Try the new version (v2.3.0) of MSIRGB.

Have just tried the new version. Works perfectly!!!! You are a god among men!
Brilliant work man, MSI needs to take note 👍

ixjf, 2020-09-05:
Great! Thanks for the feedback.

closed 2020-09-05 by ixjf