ixjf/MSIRGB #116: Hue Wheel problem

surftor, 2020-06-08:
The hue wheel goes through the colours but between each colour it transitions it to white. Any idea why/how to fix?
Thanks

ixjf, 2020-06-08:
Could be that some other program is also changing the LED colours at the same time.

ixjf, 2020-09-05:
Did you fix the issue?

surftor, 2020-09-06:
Hey sorry for not replying earlier. If it happens it is due to my PC losing
power completely. I can just reinstall mystic, boot it, then uninstall it.
Thanks!

ixjf, 2020-09-06:
Which motherboard do you have?

surftor, 2020-09-06:
B450M Gaming Plus

ixjf, 2020-09-06:
Okay, I'll look into this ASAP!

surftor, 2020-09-06:
Thanks for the dedication!

ixjf, 2021-08-23:
Sorry, is this still happening? So I know whether I should investigate for a fault on my side.

surftor, 2021-08-23:
It's all good, It just requires MSI Mystic Light to change to any profile
and then it works fine, with the RGB loop.
Thanks for your continued work on this project, it's made my system look
much better!
-surftor

ixjf, 2021-08-23:
Huh, you shouldn't have to. I'll try to figure this out this week. Sorry for the VERY LONG delay, but with university, this is not at all one of my priorities :P