ixjf/MSIRGB #122: "Failed to load driver" error.

M0drah, 2020-06-15:
I have followed the directions, I have all my folders done correctly as shown in the readme. 
When I click on "MSIRGB.exe" to run it, the User Account Control comes up, I click "yes" and almost immediately get this error: 

"Failed to load driver. This could be either due to some program interfering with MSIRGB's driver, or it could be a bug. Please report this on the issue tracker at https://github.com/ixjf/MSIRGB"

[image]

My folders:
[image]

[image]

Is there something I'm doing incorrectly?

Thanks!
M0drah

ixjf, 2020-06-15:
Likely you have another program blocking the driver from loading. Riot Vanguard and other programs that control hardware might be the root cause.

nyxmon, 2020-06-15:
This definitely sounds like Vanguard or similar "anti-cheat" program interfering. It shows the exact same error on mine when I start **MSIRGB.exe** (whilst Vanguard is running).

jsftv, 2020-06-16:
Can confirm. Riot Vanguard is blocking mine now, as well.

ixjf, 2020-09-05:
Did you fix the issue?

M0drah, 2020-09-05:
I did. It ended up being the anti-cheat for Valorant causing it. I have
since uninstalled Valorant and the software. My motherboard does not have
the colors I thought it did.

ixjf, 2020-09-05:
Alright, thanks. Vanguard is an issue I still need to handle.

closed 2020-09-05 by ixjf