ixjf/MSIRGB #123: MSI Gaming Plus Max also has the reversed color issue

psisquared2, 2020-06-17:
As the title says, the MSI Gaming Plus MAX is also affected by #106. This is in contrast to the non-MAX version which someone else reported to work without issue.

Interestingly, the MSI Mystic Light tool also applies reversed colors even after I close MSIRGB. It is only fixed after a reboot.

Please let me know if I can help by providing further information to add this mainboard to the supported list.

ixjf, 2020-09-05:
MSIRGB v2.3.0 fixes the reversed colour issue. Please try the new version.

ixjf, 2020-09-06:
Closing as it is a duplicate of #139.

closed 2020-09-06 by ixjf

psisquared2, 2020-09-09:
I can confirm that the B450 MSI Gaming Plus MAX now works as intended. Thank you!

ixjf, 2020-09-09:
Great, thanks for the feedback!