ixjf/MSIRGB #125: MSIRGB Scripts not starting on startup

AiyanH, 2020-06-21:
I have to open the MSIRGB application everytime and click the script i want. Any way so that my script runs everytime i turn on my computer? It is very annoying to run MSIRGB everytime i turn on my pc.

jacklul, 2020-06-22:
The moment you open the application and check the script you want to use it should register a Windows service that will run at boot, this will not happen if you don't run application as admin (it should auto-ask you on launch).
Make sure you do not move MSIRGB folder anywhere after doing this or it will not work, it will have to be in one place to work.

Start->Run->"services.msc" and check for "MSIRGB Script Service".

AiyanH, 2020-07-17:
Yes MSIRGB Script Service is in services.msc. The problem that is happening is if i completely shut down my computer then turn it on then msirgb doesnt run on startup. But if run the application and then restart or put computer to sleep and turn on then msirgb keeps working. The problem happens when i completely shut down and turn on my pc

ixjf, 2020-07-18:
This is a known issue. MSIRGB fails to run on a cold boot. Still don't know why. Unfortunately, I don't have access to my desktop to fix this right now, so it'll have to wait (can't tell you how long - weeks, maybe a month).

McBruno712, 2020-07-23:
What can I do if the service is not there?

ixjf, 2020-09-06:
Closing as it is a duplicate of #43. Please refer to that issue for info on the status of the fix.

closed 2020-09-06 by ixjf