ixjf/MSIRGB #126: Couldn't Open 'latest release' and 'latest example effects'

chururi, 2020-06-26:
Hi, i have a B350m Mortar Arctic. I have installed the VC Redist 2017 x64. After i downloaded the 'latest release', the (MSIRGB-v2.2.1.2.7z), after i try to click on it. This following messagge appeared.

"This file does not have an app associated with it for performing this action. Please install an app or, if one is already installed, create an association in the Default Apps Settings page."

So what is this 'App' its talking about? I have tried to download the GamingApp and MSI Command Center. But still this message appeared. What can i do for now? Will be waiting for the reply. Thanks.

ixjf, 2020-06-26:
You need a file archiver installed, like 7-Zip or WinRar, to open these kinds of files. I assumed everyone would have one installed already, but in the future, I'll try to package releases in the ZIP format that can be opened without the need for third-party applications.

chururi, 2020-06-26:
Of course it is! This PC was actually build yesterday. I still didn't have WinRAR.
Everything works perfectly, except the color isn't the same as what i input. For further info, i didn't know if the color is reversed or not. I put it as purple, but it came out as white  pinkish. Thanks again for the help, sir!

ixjf, 2020-06-26:
Which motherboard is it?

chururi, 2020-06-26:
It's B350m Mortar Arctic

ixjf, 2020-06-27:
Alright, thanks. Will look into that issue ASAP.

chururi, 2020-06-27:
Thank you so much for the support.

ixjf, 2020-09-05:
MSIRGB v2.3.0 fixes the reversed colour issue. Please try the new version.

ixjf, 2020-09-05:
Although MSIRGB v2.3.0 fixes this issue, this motherboard is not on the list of supported motherboards, hence the fix doesn't apply. I'll try to fix that soon.

ixjf, 2021-08-21:
Fixed! (or I hope so - do test to make sure).

commit 32003298 (2021-08-21): Fix #126 (experimental)
closed 2021-08-21 by ixjf