ixjf/MSIRGB #128: Spike in CPU activity when plugging in ANY USB device while MSIRGB is running

BruebachL, 2020-07-01:
Pretty self-explanatory. I've got a B350 PC Mate and whenever MSIRGB is running and I plug in or disconnect a USB device (Phone, Gamepad, USB Stick) MSIRGB spikes in CPU activity and my whole system starts freezing a little (sound distortion, mouse laggy). It will also tab me out of any fullscreen applications and return me to desktop. 
I have a Ryzen 5 3600X. What is causing this behavior?

ixjf, 2020-07-01:
I don't have my desktop with me right now (and won't have it until August) so I cannot attempt to reproduce this. If someone else could, that'd be nice.

Anyhow, are you running any particular script?

ixjf, 2020-09-05:
Did you fix this?

ixjf, 2020-09-05:
I cannot reproduce the issue.

peroanjo, 2021-02-04:
I've got the exact same issue with a 3600x too. Might be an incompatibility with this specific processor.

ixjf, 2021-02-04:
And this never happens when MSIRGB is turned off? And this is with the MSIRGB service running, the GUI, or both?

peroanjo, 2021-02-04:
It happens when the MSIRGB service is running, stopping it and restarting solved the cpu spikes.

ixjf, 2021-02-04:
Can confirm. Though I do not get any CPU usage spikes. But my mouse starts freezing, yes. No clue why.

ixjf, 2021-02-04:
... and I can't replicate it anymore.

ixjf, 2021-08-13:
Note to myself: random mouse freezes during normal Windows usage. may or may not be caused by MSIRGB.

ixjf, 2021-08-21:
Do you still have this problem? I have some random mouse lag sometimes but I've tried turning off MSIRGB and the same happens, so I don't think that has anything to do with MSIRGB.

closed 2021-09-02 by ixjf