ixjf/MSIRGB #129: It wont start

bit-greg, 2020-07-10:
So recently i wanted to install MSIRGB. I already had it working on another machine withe the same Mainboard (MSIM-Pro-VDH), but now i cant open it. As soon as I try opening it, it asks for admin rights but then it dissapears immediately. Also it doesn´t show up in Task Manager. I have no Idea why it wont work since it already worked on the same mainboard and ive VCRedist installed.
Thx for help.

ixjf, 2020-08-19:
I'm sorry for the delayed response. Did you get it fixed?

bit-greg, 2020-08-19:
Yup finally got it. There were some corrupted files in my operating system that after reinstalling windows were finally fixed. I think it was some kind of driver issue that caused the problem.

closed 2020-08-19 by bit-greg